openbmc/openbmc #2010: phosphor-inventory fails to start

shenki, 2017-07-24:
v1.99.8-71-ge319c8c

```
Jul 24 04:21:36 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:21:36 witherspoon systemd[1]: Starting Phosphor Network Manager...
Jul 24 04:21:37 witherspoon phosphor-inventory[1251]: /usr/sbin/phosphor-inventory: symbol lookup error: /usr/sbin/phosphor-inventory: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project9Inventory9Decorator6server8AssetTagE
Jul 24 04:21:37 witherspoon phosphor-settings-manager[1261]: /usr/sbin/phosphor-settings-manager: symbol lookup error: /usr/sbin/phosphor-settings-manager: undefined symbol: _ZTIN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:21:36 witherspoon systemd[1]: Stopped Phosphor Settings Daemon.
Jul 24 04:21:36 witherspoon systemd[1]: Starting Phosphor Settings Daemon...
```

williamspatrick, 2017-07-25:
This looks like a user error.  I am running this exact code on a Witherspoon and it works just fine.  The errors here indicate the wrong library is loaded.  Either you have patches on your machine, such as a libphosphor-dbus.so left over from before, or something is wrong with your build environment.

```
root@witherspoon:~# systemctl status xyz.openbmc_project.Inventory.Manager.service --no-pager
● xyz.openbmc_project.Inventory.Manager.service - Phosphor Inventory Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Inventory.Manager.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-07-25 17:05:29 UTC; 1min 52s ago
 Main PID: 858 (phosphor-invent)
   CGroup: /system.slice/xyz.openbmc_project.Inventory.Manager.service
           └─858 /usr/sbin/phosphor-inventory

Jul 25 17:05:23 witherspoon systemd[1]: Starting Phosphor Inventory Manager...
Jul 25 17:05:29 witherspoon systemd[1]: Started Phosphor Inventory Manager.
```

closed 2017-07-25 by williamspatrick

shenki, 2017-07-26:
This build was downloaded from the Jenkins openpower.xyz