openbmc/openbmc #2011: xyz.openbmc_project.Common.Error.InternalFailure on boot to Standby [ Witherspoon]

gkeishin, 2017-07-24:
1. What is this error about? 
2.  Could there be additional data to tell what failure is this about?

Driver info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-71"
VERSION_ID="v1.99.8-71-ge319c8c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-71"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-d6ae40dc4c4dff3265cc254d404ed6b03fcc2206 #1 Mon Jul 24 03:34:34 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 17:34:33 up 5 min,  load average: 0.17, 0.67, 0.39
root@witherspoon:~#
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#
```

```
$ ./rest_cmd -i xx.xx.xx.xx -o esel
Fetching esel from BMC
Error entry(s) found:
[u'/xyz/openbmc_project/logging/entry/2', u'/xyz/openbmc_project/logging/entry/1']
Entry : /xyz/openbmc_project/logging/entry/2
{
    "AdditionalData": [],
    "Id": 2,
    "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1492709439401,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [],
    "Id": 1,
    "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1492709413782,
    "associations": []
}

```

- [x] Journald 
[Journald_v1.99.8-71-ge319c8c_comm_failure.txt](https://github.com/openbmc/openbmc/files/1168839/Journald_v1.99.8-71-ge319c8c_comm_failure.txt)

geissonator, 2017-07-24:
Looks to be phosphor-network-manager related:
```
Apr 20 17:30:39 witherspoon phosphor-network-manager[995]: Failed to get hostname
Apr 20 17:30:39 witherspoon phosphor-network-manager[995]: The operation failed internally.
Apr 20 17:30:39 witherspoon phosphor-log-manager[926]: Failed to find metadata
Apr 20 17:30:39 witherspoon dbus[920]: [system] Failed to activate service 'org.freedesktop.hostname1': timed out
Apr 20 17:30:39 witherspoon systemd[1]: Started Phosphor Network Manager.
Apr 20 17:30:39 witherspoon systemd[1]: Reached target Phosphor System - Standby.
```

ratagupt, 2017-07-24:
This one is related to 1949
Apr 20 17:30:39 witherspoon phosphor-network-manager[995]: Failed to get hostname
Apr 20 17:30:39 witherspoon phosphor-network-manager[995]: The operation failed internally.

Gerrit fix is:  https://gerrit.openbmc-project.xyz/#/c/5398/

gkeishin, 2017-07-27:
The #1949 and this one seems to not 1x1 related. I still see this error on latest build
```
Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [],
    "Id": 1,
    "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501010668392,
    "associations": []
}
```

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-98"
VERSION_ID="v1.99.8-98-g06734f3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-98"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-bd1d1e81e194d143bf6a1681ab35d44f77a9db10 #1 Thu Jul 27 02:33:28 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: ConfigParser: Section not found
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: The operation failed internally.
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: ConfigParser: Section not found
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: The operation failed internally.
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: ConfigParser: Key not found
Jul 25 19:24:19 witherspoon phosphor-network-manager[992]: The operation failed internally.
```

rfrandse, 2017-08-02:
<https://gerrit.openbmc-project.xyz/5736> Don't commit the error during getting of DHCP value
Resolves  #2011 xyz.openbmc_project.Common.Error.InternalFailure on boot to Standby [ Witherspoon]

closed 2017-08-10 by williamspatrick