openbmc/openbmc #2012: Error while setting time

trajeswaran, 2017-07-24:
While setting time, I got the below error. This error is not always reproducible. But, once this error occurs, then subsequent attempts to set time results in the same error. 

sptime -timeofday 20170703035900
***ERROR: BMCSetSystemTime : JSON response is illformed.
***ERROR: {
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Error setting time on BMC',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 721, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 778, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 956, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 930, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 895, in wrap", 
      "    data = callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 829, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 860, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 183, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 318, in do_post", 
      "    return request.route_data['method'](*request.parameter_list)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.Failed: Error setting time on BMC"
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}***ERROR: cmtRtimSetTimeOfDay : BMCSetSystemTime() failed.

======================================

Journalctl  :

Jul 03 03:50:46 witherspoon timemanager[979]: Request to set BMC time
Jul 03 03:50:46 witherspoon timemanager[979]: Curr_Mode: MANUAL Curr_Owner: BMC
Jul 03 03:50:46 witherspoon timemanager[979]:  BMC TIME : 2017-07-03 03:56:00
Jul 03 03:50:46 witherspoon dbus[866]: [[0;1;39m[system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus
-org.freedesktop.timedate1.service'[[0m
Jul 03 03:50:46 witherspoon systemd[1]: Starting Time & Date Service...
Jul 03 03:50:46 witherspoon dbus[866]: [[0;1;39m[system] Successfully activated service 'org.freedesktop.timedate1'[[0m
Jul 03 03:50:46 witherspoon systemd-timedated[2836]: [[0;1;39m/etc/localtime should be a symbolic link to a time zone data file in /
usr/share/zoneinfo/.[[0m
Jul 03 03:50:46 witherspoon systemd[1]: Started Time & Date Service.
Jul 03 03:50:46 witherspoon timemanager[979]: Error: Operation already in progresssetting time on BMC
Jul 03 03:50:46 witherspoon dbus[866]: [[0;1;39m[system] Rejected send message, 0 matched rules; type="error", sender=":1.66" (uid=0
 pid=979 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requeste
d_reply="0" destination=":1.69" (uid=0 pid=996 comm="python /usr/sbin/phosphor-gevent rest_dbus ")[[0m
Jul 03 03:50:47 witherspoon phosphor-gevent[996]: Traceback (most recent call last):
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 862, in _handle
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:     return route.call(**args)
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 721, i
n __call__
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:     return self.callback(*a, **kw)
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 778, i
n wrap
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:     return callback(*a, **kw)
Jul 03 03:50:47 witherspoon phosphor-gevent[996]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 956, i
n __call__

mine260309, 2017-07-24:
The log seems to show that the time is set twice, and the second try results in `Operation already in progress` error.

vishwabmc, 2017-07-25:
@trajeswaran Could you use the latest build ?. I have improved the error message / exception in one of the commits https://gerrit.openbmc-project.xyz/#/c/5117/ that got merged couple days ago.

With this, we get exact systemd exception message and will help isolate.

trajeswaran, 2017-07-26:
With the latest build, I am getting this error.  Please find attached the journal entries.
[journal.txt](https://github.com/openbmc/openbmc/files/1176332/journal.txt)

Jul 26 10:10:05 witherspoon phosphor-gevent[1168]: ::8029:6876:8e91:f402%1995647772 - - [2017-07-26 10:10:05] "POST /login HTTP/1.1"
 200 320 0.033275
Jul 26 10:10:05 witherspoon timemanager[1126]: Request to set BMC time
Jul 26 10:10:05 witherspoon timemanager[1126]: Curr_Mode: MANUAL Curr_Owner: BMC
Jul 26 10:10:05 witherspoon timemanager[1126]:  BMC TIME : 2017-07-26 11:09:53
Jul 26 10:10:05 witherspoon dbus[987]: [[0;1;39m[system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus
-org.freedesktop.timedate1.service'[[0m
Jul 26 10:10:05 witherspoon systemd[1]: Starting Time & Date Service...
Jul 26 10:10:05 witherspoon dbus[987]: [[0;1;39m[system] Successfully activated service 'org.freedesktop.timedate1'[[0m
Jul 26 10:10:05 witherspoon systemd[1]: Started Time & Date Service.
Jul 26 10:10:05 witherspoon systemd-timedated[1193]: [[0;1;39m/etc/localtime should be a symbolic link to a time zone data file in /
usr/share/zoneinfo/.[[0m
Jul 26 10:10:05 witherspoon dbus[987]: [[0;1;39m[system] Rejected send message, 0 matched rules; type="error", sender=":1.105" (uid=
0 pid=1126 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.timedate1.AutomaticTimeSy
ncEnabled" requested_reply="0" destination=":1.107" (uid=0 pid=1168 comm="python /usr/sbin/phosphor-gevent rest_dbus ")[[0m
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]: Traceback (most recent call last):
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 862, in _handle
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return route.call(**args)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 759, 
in __call__
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return self.callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 816, 
in wrap
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 999, 
in __call__
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return self.callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 973, 
in wrap
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return self.to_jsonp(callback(*a, **kw))
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 1734, in wrapper
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     rv = callback(*a, **ka)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 938, 
in wrap
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     data = callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 867, 
in wrap
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 898,
in wrap
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return callback(*a, **kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 184, 
in __call__
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return getattr(self, 'do_' + request.method.lower())(**kw)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 319, 
in do_post
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     return request.route_data['method'](*request.parameter_list)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     **keywords)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_b
locking
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]:     message, timeout)
Jul 26 10:10:05 witherspoon phosphor-gevent[1168]: DBusException: org.freedesktop.timedate1.AutomaticTimeSyncEnabled: Automatic time
 synchronization is enabled
Jul 26 10:10:06 witherspoon phosphor-gevent[1168]: ::8029:6876:8e91:f402%1995647772 - - [2017-07-26 10:10:05] "POST //org/openbmc/Ti
meManager/action/SetTime HTTP/1.1" 500 2406 0.746083
Jul 26 10:10:18 witherspoon avahi-daemon[970]: [[0;1;39mService name conflict for "sftp" (/services/sftp.service), retrying with "sf
tp #19".[[0m
Jul 26 10:10:18 witherspoon avahi-daemon[970]: [[0;1;39mService name conflict for "phosphor_rest" (/services/phosphor_rest.service),
 retrying with "phosphor_rest #19".[[0m
Jul 26 10:10:18 witherspoon avahi-daemon[970]: [[0;1;39mService name conflict for "obmc_console" (/services/obmc_console.service), r
etrying with "obmc_console #19".[[0m

vishwabmc, 2017-07-26:
** #STATUS **

trajeswaran, 2017-09-11:
After updating the MFG tool to use the Rest API /xyz/openbmc_project/time/sync_method to set the mode and the Rest API /xyz/openbmc_project/time/owner to set the owner, I am not getting this problem. So, this defect can be closed.

trajeswaran, 2017-09-11:
This fix will be available in the next level of toolkit release to MFG.

closed 2017-09-11 by trajeswaran