openbmc/openbmc #2013: Host does not come up after PNOR code update

charleshofer, 2017-07-24:
After a PNOR code update via REST, the host does not come up after a `obmcutil poweron`. The host serial console remains empty. Journal is attached. Seems mboxd related.

[journal.txt](https://github.com/openbmc/openbmc/files/1170492/journal.txt)

charleshofer, 2017-07-24:
Version information:

```
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-69"
VERSION_ID="v1.99.8-69-gf930045-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-69"
BUILD_ID="v1.99.8"
```

dkodihal, 2017-07-24:
A code update from an image not supporting the virtual PNOR to one supporting it would need to be disruptive. So for example a BMC reboot would fix this.

charleshofer, 2017-07-24:
Did the following:
1. Uploaded a fresh BMC image (just to be doubly sure we're on the right one)
2. Did a PNOR code update
3. Rebooted
4. Did a second PNOR code update
Still seeing the issue. Uploaded the journal.

[journal.txt](https://github.com/openbmc/openbmc/files/1171444/journal.txt)

geissonator, 2017-07-25:
We're hoping moving the mbox reset to power on in #1915 will fix this.

amboar, 2017-07-27:
This is in the journal:

```
Jul 19 17:45:21 witherspoon openpower-update-manager[954]: Error PNOR current version is empty
Jul 19 17:45:21 witherspoon mboxd[957]: The operation failed internally.
Jul 19 17:45:21 witherspoon mboxd[957]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 19 17:45:21 witherspoon mboxd[957]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Jul 19 17:45:21 witherspoon mboxd[957]: [ 1500486321.363449782] Partition HBB not found
```

Sure, there's an mboxd error. I'm not convinced it should crash in this circumstance, but that's a separate conversation. There's also a openpower-update-manager error. It looks like a bad PNOR image or missing mount points to me.

amboar, 2017-07-27:
@geissonator if moving the reset fixes the problem, then we're doing updates wrong. The update process needs to either invoke `mboxctl` or issue the right commands on DBus to `mboxd`. Specifically, we need to issue `mboxctl --suspend` before touching the PNOR, and `mboxctl --resume modified` once finished (or, perform the equivalent DBus commands)

closed 2017-09-26 by charleshofer