openbmc/openbmc #2014: Priority disappears from enumerate, and RedundencyPriority disappears from D-Bus after PNOR image upload

charleshofer, 2017-07-24:
About 5 minutes after a PNOR image upload the Priority attribute disappears from the image version object in the REST interface and the RedundencyPriority object is missing from D-Bus.

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-69"
VERSION_ID="v1.99.8-69-gf930045-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-69"
BUILD_ID="v1.99.8"
```

[journal.txt](https://github.com/openbmc/openbmc/files/1170801/journal.txt)

saqibkh, 2017-07-24:
@charleshofer This is happening because of a core-dump from mboxd.
This shouldn't happen if you reboot the BMC after the first successful PNOR Update.
Let me know if you still see an issue after rebooting the BMC.
Thanks

charleshofer, 2017-07-24:
Did the following
1. Updated the BMC image, just to make sure
2. Did a PNOR update
3. Rebooted
4. Did another PNOR update
The problem persists

charleshofer, 2017-07-24:
Journal in case you're interested.

[journal.txt](https://github.com/openbmc/openbmc/files/1171451/journal.txt)

closed 2017-07-28 by charleshofer