openbmc/openbmc #2017: Image uploads with invalid purposes are not deleted from the /tmp/images directory

charleshofer, 2017-07-24:
When uploading a PNOR image with an invalid Purpose through REST and TFTP, the image is not removed from the /tmp/images directory.

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-69"
VERSION_ID="v1.99.8-69-gf930045-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-69"
BUILD_ID="v1.99.8"
```

[journal.txt](https://github.com/openbmc/openbmc/files/1171182/journal.txt)

saqibkh, 2017-07-24:
@charleshofer 
Can you paste the MANIFEST's purpose here so we can recreate this.
Also post the enumerate listing before and after uploading the image. 
Thanks

charleshofer, 2017-07-24:
Enumerate does not change for bad manifest purposes. The BMC, correctly, does not create an entry in enumerate. Here's the MANIFEST contents though.
```
purpose=thisisaboguspurpose
version=open-power-witherspoon-v1.16-53-gfb396b3-dirty-badmanifest
extended_version=buildroot-2017.02.2-7-g23118ce,skiboot-f0ebbc8,hostboot-56a7aab,linux-4.11-openpower1-pd4d4971,witherspoon-xml-cee25d2-pac01485,occ-43b9907,hostboot-binaries-8cbbc67,capp-ucode-9c73e9f,sbe-9b98de3-p43ac507
```

charleshofer, 2017-07-24:
Enumerate just in case, but it stays the same:
```
[Charles.Hofer@oc4481562561 ~]$ bmc-enumerate 9.3.21.43
{
  "data": {
    "/xyz/openbmc_project/software/3c4bb092": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "v1.99.8-69-gf930045-dirty"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

saqibkh, 2017-07-24:
I can see the software version with the wrong purpose in enumerate with the purpose set to Unknown.
Perhaps @gtmills can tell us if the files under /tmp/images should be deleted automatically here.

-bash-4.1$ ./rest_cmd -i 9.3.181.15 -o GET -u  /xyz/openbmc_project/software/enumerate
Executing: GET  /xyz/openbmc_project/software/enumerate

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/3c4bb092": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-69-gf930045-dirty"
    },
    "/xyz/openbmc_project/software/d53b7ca4": {
        "Path": "/tmp/images/d53b7ca4",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Unknown",
        "Version": "open-power-witherspoon-v1.16-53-gfb396b3-dirty-badmanifest"
    }
}

mtritz, 2018-01-08:
Is this actually a bug? We handle bad or missing purpose when checking the manifest here: 
https://github.com/openbmc/phosphor-bmc-code-mgmt/blob/master/image_manager.cpp#L114

gtmills, 2018-01-08:
Yes, the image manager was designed to set the purpose to unknown and keep going if the purpose is invalid. 
The image manager could be changed to not allow invalid purposes by changing the catch block to 
```
        log<level::ERR>("Error invalid purpose");
        report<ManifestFileFailure>(ManifestFail::PATH(manifestPath.c_str()));
        return -1;
```
https://github.com/openbmc/phosphor-bmc-code-mgmt/blob/master/image_manager.cpp#L130

but is this what we want? why have the unknown purpose then?

anoo1, 2018-01-08:
Closing. The manifest can have any text on the Purpose field, since support for other firmware images (like PCIe cards) can be added and the image should be a software object in the image manager even if there's not an updater process for it. Only if the image doesn't have a manifest, it'd be deleted.

closed 2018-01-08 by anoo1