openbmc/openbmc #2019: ObjectMapper: Unknown lvalue '--blacklists' in section 'Service'

shenki, 2017-07-25:
Seen at boot with v1.99.8-41-g86a4abc

```
    8.580000] systemd[1]: systemd 232 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN)
[    8.580000] systemd[1]: Detected architecture arm.
[    8.770000] systemd[1]: Set hostname to <spoon1-bmc>.
[   10.130000] systemd[1]: [/lib/systemd/system/xyz.openbmc_project.ObjectMapper.service:9] Unknown lvalue '--blacklists' in section 'Service'
[   10.300000] systemd[1]: Listening on Syslog Socket.
[   10.420000] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[   10.460000] systemd[1]: Listening on Journal Socket (/dev/log).
[   10.500000] systemd[1]: Listening on udev Control Socket.
[   10.540000] systemd[1]: Listening on Network Service Netlink Socket.
[   10.580000] systemd[1]: Reached target Swap.
[   13.920000] systemd-journald[655]: Received request to flush runtime journal from PID 1

```

geissonator, 2017-07-25:
@gkeishin fixed this one for us a few days ago, George could you let us know the issue number and close this one out?

gkeishin, 2017-07-25:
Already fixed in https://github.com/openbmc/openbmc/issues/1997

Available fixed in latter builds

closed 2017-07-25 by gkeishin