openbmc/openbmc #2020: phosphor-fan-control throwing exception

shenki, 2017-07-25:
phosphor-fan-control needs some rework so it doesn't crash.

It also needs a better error message than "xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally."

Educated users will not have a chance of diagnosing that.

```
Jul 25 04:03:49 spoon1-bmc systemd[1]: Starting Start Power0...
Jul 25 04:03:51 spoon1-bmc phosphor-fan-control[1092]: Error in mapper call to get service name
Jul 25 04:03:51 spoon1-bmc phosphor-fan-control[1092]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 25 04:03:51 spoon1-bmc phosphor-fan-control[1092]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Jul 25 04:03:51 spoon1-bmc phosphor-fan-control[1092]: The operation failed internally.
Jul 25 04:03:51 spoon1-bmc systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Main process exited, code=dumped, status=6/ABRT
Jul 25 04:03:51 spoon1-bmc systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Unit entered failed state.
Jul 25 04:03:51 spoon1-bmc systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Failed with result 'core-dump'.
```

spinler, 2017-07-25:
@shenki For better or for worse, what we need to figure this out is in the journal metadata of the 'error in mapper call to get service name'.

Can you either run a
```
 journalctl --no-pager MESSAGE='Error in mapper call to get service name' -o verbose
```
or tell me the IP address of this system so I can take a look.

shenki, 2017-07-25:
[verbose-error-output.txt](https://github.com/openbmc/openbmc/files/1173457/verbose-error-output.txt)

shenki, 2017-07-25:
The system is on an internal network that requires a few extra hoops to jump through. If you still need access please ping me, and we can sort that out for you.

spinler, 2017-07-25:
```
    PATH=/xyz/openbmc_project/sensors/fan_tach/fan2_0
    INTERFACE=xyz.openbmc_project.Control.FanSpeed
```
Can you check if there is a phosphor-hwmon-readd instance running for the max31785 chip?

shenki, 2017-07-26:
This suggests no?

```
root@spoon1-bmc:~# ps fax | grep phosphor-hwmon-readd
 2810 root       0:23 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@70
 2816 root       0:24 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@71
 2818 root       0:23 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@71
 2819 root       0:23 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@70
```

spinler, 2017-07-26:
Yea, it isn't.  That is why fan control is failing.  Has nothing to do with the level of the MAX31785.

msbarth, 2017-10-10:
A similar issue is being tracked under #2420 which was against a more recent build of fan control.

closed 2017-10-10 by spinler