openbmc/openbmc #2021: witherspoon: host fails to start

shenki, 2017-07-25:
Running v1.99.8-41-g86a4abc on our pass1 witherspoon:

```
Jul 25 04:03:42 spoon1-bmc phosphor-host-state-manager[1030]: Host State transaction request
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-avsbus\x2dworkaround.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-vrm\x2dcontrol.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-fsi\x2dscan.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Starting Witherspoon Fan Watchdog Reset...
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-start_host.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-vcs_off.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Starting Wait for /xyz/openbmc_project/watchdog/host0...
Jul 25 04:03:42 spoon1-bmc systemd[1]: Starting Wait for /org/openbmc/control/chassis0...
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-vcs_on.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-ir35221\x2don\x2dbind.slice.
Jul 25 04:03:42 spoon1-bmc systemd[1]: Created slice system-ir35221\x2don\x2dunbind.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-fsi\x2dbind.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-op\x2dwait\x2dpower\x2don.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Starting Wait for Power0 to turn on...
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-mapper\x2dsubtree\x2dremove.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-pcie\x2dslot\x2ddetect.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-op\x2dpower\x2dstart.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-avsbus\x2ddisable.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-phosphor\x2dwatchdog.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Started Phosphor poweron watchdog.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Starting Wait for /xyz/openbmc_project/led/groups/power_on...
Jul 25 04:03:43 spoon1-bmc systemd[1]: Reached target Powerwitherspoon-fan-watchdog (On).
Jul 25 04:03:43 spoon1-bmc systemd[1]: Starting power on mode unbind ir35221 device driver devs...
Jul 25 04:03:43 spoon1-bmc systemd[1]: Started Witherspoon Fan Watchdog Daemon.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Started Phosphor Fan Control Initialization.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-cfam_override.slice.
[  607.440000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-ucd_disable_vcs.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Starting Disable UCD VCS rails...
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-avsbus\x2denable.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-phosphor\x2dgpio\x2dmonitor.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Started Phosphor GPIO checkstop monitor.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Reached target Start Host checkstop monitor.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-phosphor\x2dfan\x2dpresence\x2dtach.slice.
Jul 25 04:03:43 spoon1-bmc systemd[1]: Created slice system-vcs_workaround.slice.
Jul 25 04:03:44 spoon1-bmc kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:03:44 spoon1-bmc kernel[915]: [  607.440000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:03:44 spoon1-bmc systemd[1]: Started Witherspoon Fan Watchdog Reset.
Jul 25 04:03:44 spoon1-bmc systemd[1]: Started Wait for /org/openbmc/control/chassis0.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Started Wait for /xyz/openbmc_project/led/groups/power_on.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Started power on mode unbind ir35221 device driver devs.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Stop job pending for unit, delaying automatic restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Stop job pending for unit, delaying automatic restart.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:03:45 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 04:03:45 spoon1-bmc systemd[1]: Stopping Phosphor Hwmon Poller...
[... 431 lines cut ...]
Jul 25 04:04:06 spoon1-bmc power_control.exe[1031]: PowerControl: pgood: 0, setting reset BMC_CP0_RESET_N to 0
Jul 25 04:04:06 spoon1-bmc system_manager.py[988]: Running System State: HOST_POWERED_OFF
Jul 25 04:04:06 spoon1-bmc timemanager[1056]:  PGOOD has changed..
Jul 25 04:04:06 spoon1-bmc timemanager[1056]: Changing time settings allowed now
Jul 25 04:04:06 spoon1-bmc timemanager[1056]:  User Settings have changed..
Jul 25 04:04:06 spoon1-bmc systemd[1]: Started Wait for Power0 to turn off.
Jul 25 04:04:06 spoon1-bmc systemd[1]: Starting Reload mboxd after power off...
Jul 25 04:04:06 spoon1-bmc systemd[1]: Starting Disable VCS on a power off...
Jul 25 04:04:07 spoon1-bmc mboxctl[1437]: Reset: Success
Jul 25 04:04:07 spoon1-bmc systemd[1]: Started Reload mboxd after power off.
Jul 25 04:04:07 spoon1-bmc systemd[1]: Started Disable VCS on a power off.
Jul 25 04:04:09 spoon1-bmc phosphor-ledmanager[932]: De-Asserting LEDs
Jul 25 04:04:09 spoon1-bmc phosphor-ledmanager[932]: No service providers for physical LED
[... cut ...]
Jul 25 04:04:10 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:04:10 spoon1-bmc systemd[1]: Reached target Chassis0 (Power Off).
Jul 25 04:04:10 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:04:10 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:04:10 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:04:10 spoon1-bmc phosphor-chassis-state-manager[1027]: Received signal that power OFF is complete
Jul 25 04:04:10 spoon1-bmc phosphor-chassis-state-manager[1027]: Change to Chassis Power State
```

shenki, 2017-07-25:
The same on today's master (v1.99.8-79-g246c6f9):

```
Jul 25 04:59:22 spoon1-bmc phosphor-host-state-manager[1040]: Host State transaction request
Jul 25 04:59:22 spoon1-bmc systemd[1]: Started Phosphor poweron watchdog.
Jul 25 04:59:22 spoon1-bmc systemd[1]: Started Phosphor GPIO checkstop monitor.
Jul 25 04:59:22 spoon1-bmc systemd[1]: Starting power on mode unbind ir35221 device driver devs...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Starting Witherspoon Fan Watchdog Reset...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Starting Disable UCD VCS rails...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Chassis0 (Power Off).
Jul 25 04:59:23 spoon1-bmc systemd[1]: Reached target Start Host checkstop monitor.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Reload mboxd after power off.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Power off mode bind ir35221 device driver devs.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Disable VCS on a power off.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Power0 Off.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Wait for Power0 to turn off.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Stop Power0.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Power0 Off (Pre).
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Host0 (Stopped).
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Host0 (Stopping).
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Power off mode unbind ir35221 device driver devs.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Stop Host0 (Pre).
Jul 25 04:59:23 spoon1-bmc systemd[1]: Starting Assert power_on LED...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped Assert power_on LED.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Starting Wait for Power0 to turn on...
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Stop Host0.
Jul 25 04:59:23 spoon1-bmc systemd[1]: Stopped target Stop Host checkstop monitor.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Started power on mode unbind ir35221 device driver devs.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Started Witherspoon Fan Watchdog Reset.
Jul 25 04:59:24 spoon1-bmc dropbear[1037]: Exit (root): Disconnect received
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Stop job pending for unit, delaying automatic restart.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
[  501.360000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:59:24 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 04:59:25 spoon1-bmc kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:59:25 spoon1-bmc kernel[945]: [  501.360000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 04:59:24 spoon1-bmc systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:59:24 spoon1-bmc systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 04:59:24 spoon1-bmc systemd[1]: Stopping Phosphor Hwmon Poller...
[... cut ...]
Jul 25 04:59:52 spoon1-bmc systemd[1]: Started Power off mode bind ir35221 device driver devs.
Jul 25 04:59:53 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:59:53 spoon1-bmc systemd[1]: Started Phosphor Hwmon Poller.
Jul 25 04:59:53 spoon1-bmc systemd[1]: Reached target Chassis0 (Power Off).
Jul 25 04:59:53 spoon1-bmc phosphor-chassis-state-manager[1045]: Received signal that power OFF is complete
Jul 25 04:59:53 spoon1-bmc phosphor-chassis-state-manager[1045]: Change to Chassis Power State
```

spinler, 2017-07-25:
#2020 is why this doesn't power on.  closing this.

closed 2017-07-25 by spinler