openbmc/openbmc #2023: Networkd:- NCSI driver support is needed for supporting VLAN on OpenBmc

ratagupt, 2017-07-25:
Here is the background discussion.
```>
> OpenBMC has requirement to support the VLAN,After doing the necessary VLAN
> configurations we observed that the openbmc/ncsi chip is not sending the
> tagged packet out.
>
> On the other side,if I ping the openbmc machine from some other linux
> machine
> then tagged packet doesn't reach to openbmc machine.
>
> In order to simplify the testing setup,We made a direct connection with some
> linux machine to openbmc.
>
> Scenario 1: Ping test from test/linux machine to the openbmc machine.
> ========
> Observation:- We captured the packet on the test machine which shows that
> packet
> were going out from the test machine and it was having 8021.q tag
> but it was not reaching to openbmc machine.
>
> Scenario 2: Ping test from openbmc machine to test/linux machine
> ========
> Observation:- Captured packet shows that packet were going out
> from the openbmc machine and it was having 8021.q tag but it was
> not reaching to test machine.
>
> Note:- We made sure that packet were going out from the VLAN interface.
>
> Is it related to the NCSI driver? Do we have NCSI driver support on openbmc
> to create the
> VLAN.

Gavin's reply on another thread:

> Currently, NCSI stack doesn't support VLAN. It needs some code to
> support and we need someone to work on it.
>
> There has code supporting VLAN filters. With it, the NCSI channel is
> associated with one or multiple VLANs. However, we need add code in
> ftgmac100.c to support VLAN based ethernet devices.

Benjamin's  reply
> >I added vlan support to ftgmac100 already, we need to plumb in the
> >NCSI bits (or if they are there, hook them up with ftgmac100 properly).

Cheers,
Ben.
```

pradeeps50, 2017-07-31:
Would it be worth trying by turning off checksum offload? If that works may provide clues

sammj, 2017-08-10:
RFC patches are here:
http://patchwork.ozlabs.org/patch/799576/
http://patchwork.ozlabs.org/patch/799577/
http://patchwork.ozlabs.org/patch/799578/

I'll follow this up with a upstreamable series once I get some testing results back.

pradeeps50, 2017-08-12:
In trying to be useful, just looked at the NCSI spec. So, it is very new to me. Others have already commented about the patch, so I will skip those.

From the spec:

When checking VLAN field values, the Network Controller shall match against the enabled VLAN Tag Filter values that were configured with the Set VLAN Filter command. The Network Controller shall also match on the TPI value of 8100h, as specified by IEEE 802.1q.

ftgmac100_rx_packet() doesn't seem to check for the TPI value, or has that already been done in hardware?

Will add comments (if any) as I learn more

sammj, 2017-08-14:
Thanks for having a look @pradeeps50. I believe the ftgmac100 already supports HW VLAN filtering so we don't need to touch that driver much at all. The NCSI code is to tell the current NCSI channel to filter out certain VLAN-tagged frames as management traffic - we don't need to implement the actual VLAN tagging.

For reference the latest patches are here:
http://patchwork.ozlabs.org/patch/801068/
http://patchwork.ozlabs.org/patch/801065/
http://patchwork.ozlabs.org/patch/801067/

geissonator, 2017-08-29:
Any reason we haven't closed this issue?

closed 2017-08-30 by zahrens