openbmc/openbmc #2024: openpower-occ-control crashes on single socket system

geissonator, 2017-07-25:
I was doing some testing on zaius3.  With this system, there are 2 cpu inventory objects created, but only the first is marked present by the host.  The occ_manager.hpp code needs to either only fire on a Present:1 setting or query for it during object creation.

```
[andrewg@gfwa123 ~]$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://${BMC_IP}/xyz/openbmc_project/inventory/enumerate
{
  "data": {
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "", 
      "Cached": 1, 
      "FieldReplaceable": 1, 
      "Functional": 1, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01HL329", 
      "Present": 1, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "YA3933749625", 
      "Version": "10"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1": {
      "Functional": 1, 
      "Present": 0, 
      "PrettyName": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

This was the error we hit:
```
[zaius]# journalctl -o json-pretty MESSAGE="Opening OCC device failed." | cat
{
	"__CURSOR" : "s=ecda2d4aad2c43cc9462c6dda3c7ead6;i=6de;b=e34ae4619a7546d7b90db4230df2e83d;m=13a24cb7;t=54d9c8de81dc5;x=c25ad4afeb28bea9",
	"__REALTIME_TIMESTAMP" : "1492709674589637",
	"__MONOTONIC_TIMESTAMP" : "329403575",
	"_BOOT_ID" : "e34ae4619a7546d7b90db4230df2e83d",
	"PRIORITY" : "6",
	"_MACHINE_ID" : "8e74193961d64b8883d932d32e0db904",
	"_HOSTNAME" : "zaius",
	"_UID" : "0",
	"_GID" : "0",
	"_SYSTEMD_SLICE" : "system.slice",
	"_TRANSPORT" : "journal",
	"_CAP_EFFECTIVE" : "3fffffffff",
	"CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
	"CODE_LINE" : "100",
	"CODE_FUNC" : "helper_log",
	"MESSAGE" : "Opening OCC device failed.",
	"TRANSACTION_ID" : "1677573196",
	"CALLOUT_ERRNO" : "2",
	"CALLOUT_DEVICE_PATH" : "/dev/occ2",
	"SYSLOG_IDENTIFIER" : "openpower-occ-control",
	"_PID" : "991",
	"_COMM" : "openpower-occ-c",
	"_EXE" : "/usr/sbin/openpower-occ-control",
	"_CMDLINE" : "/usr/sbin/openpower-occ-control",
	"_SYSTEMD_CGROUP" : "/system.slice/org.open_power.OCC.Control.service",
	"_SYSTEMD_UNIT" : "org.open_power.OCC.Control.service",
	"_SYSTEMD_INVOCATION_ID" : "db6699d62888467e93ccc1854f06575e",
	"_SOURCE_REALTIME_TIMESTAMP" : "1492709674587292"
}

[zaius]# ls /dev/occ*
/dev/occ1

```

geissonator, 2017-07-25:
@dkodihal - assigning to you for now since I know you're working on some code in this area (regeneration of occ objects on reset).  Maybe you can squeeze in there, otherwise we'll need to find an owner.

vishwabmc, 2017-08-11:
# **STATUS**

 - 8/11:
  - Read through the defect and the inventory object hierarchy and have identified the solution

amboar, 2017-08-13:
> Read through the defect and the inventory object hierarchy and have identified the solution

@vishwabmc This is a mistake that enrages people all over the internet. If you are going to note that you've identified the solution, please explain what the solution is and why it solves the problem.

I know you've updated your comment with some progress and have explained your approach, but please add the explanation *at the same time* as the note that you've found a solution.

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6161> Only consider CPUs that are Present and Functional
Fixes  #2024 openpower-occ-control crashes on single socket system

closed 2017-08-25 by williamspatrick