openbmc/openbmc #2025: BMC requires reboot to find OCC objects

mmpachec, 2017-07-25:
After swapping processor on one system and  flashing Attribute Overrides on different system cannot read OCC objects. Eddie James helped debug and after a reboot can scan and find the objects.

> root@witherspoon:~# busctl tree | grep -i OCC
>         ├─/org/freedesktop/systemd1/unit/org_2eopen_5fpower_2eOCC_2eControl_2eservice
>         │ └─/org/openbmc/sensors/host/cpu0/OccStatus
>         │ └─/org/openbmc/sensors/host/cpu1/OccStatus
> Service org.open_power.OCC.Control:
> Failed to introspect object / of service org.open_power.OCC.Control: Connection timed out

after reboot

> root@witherspoon:~# busctl tree | grep -i OCC
>         ├─/org/freedesktop/systemd1/unit/org_2eopen_5fpower_2eOCC_2eControl_2eservice
>         │ └─/org/openbmc/sensors/host/cpu0/OccStatus
>         │ └─/org/openbmc/sensors/host/cpu1/OccStatus
> Service org.open_power.OCC.Control:
>       ├─/org/open_power/control/occ0
>       └─/org/open_power/control/occ1

eddiejames, 2017-07-26:
Can be assigned to me. Probably a dup of #1916 or at least will be fixed when that is fixed.

gtmills, 2018-10-12:
@eddiejames #1916 is closed. Can this be closed?

closed 2019-05-14 by stale[bot]