openbmc/openbmc #2026: obmc-chassis-poweron@0.target failed [Witherspoon]

gkeishin, 2017-07-25:
`Test steps`
- [x] Boot to OS
- [x] issue `reboot`  in loops

`System goes quiesce`
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced

```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-79"
VERSION_ID="v1.99.8-79-g246c6f9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-79"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-d6ae40dc4c4dff3265cc254d404ed6b03fcc2206 #1 Tue Jul 25 01:32:57 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x0280d000..0x0280e000 !
[                                                  ] 0%IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.80
        op-build-v1.17-200-gc5f1fc5
        buildroot-2017.05-93-g9b91216
        skiboot-v5.7-rc2
        hostboot-76ac1bc
        linux-4.12.1-openpower1-p1d60f80
        petitboot-v1.5.0-pdd07c9b
        machine-xml-5575e7e
        occ-2f3397d
        hostboot-binaries-c24fccd
        capp-ucode-9c73e9f
        sbe-02021c6-pa23442d
[==================================================] 100%
root@witherspoon:~#

```

```
Apr 20 18:52:48 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Apr 20 18:52:48 witherspoon phosphor-ledmanager[944]: De-Asserting LEDs
Apr 20 18:52:48 witherspoon phosphor-ledmanager[944]: No service providers for physical LED
Apr 20 18:52:50 witherspoon systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Apr 20 18:52:50 witherspoon systemd[1]: Failed to start Run VCS workaround on host0.
Apr 20 18:52:50 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Apr 20 18:52:50 witherspoon systemd[1]: Dependency failed for Start Host0.
Apr 20 18:52:50 witherspoon systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Apr 20 18:52:50 witherspoon systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Apr 20 18:52:50 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Apr 20 18:52:50 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Apr 20 18:52:50 witherspoon systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Apr 20 18:52:50 witherspoon systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'.

Apr 20 18:52:51 witherspoon systemd[1]: pcie-slot-detect@0.service: Main process exited, code=killed, status=15/TERM
Apr 20 18:52:51 witherspoon systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Apr 20 18:52:51 witherspoon systemd[1]: pcie-slot-detect@0.service: Unit entered failed state.
Apr 20 18:52:51 witherspoon systemd[1]: pcie-slot-detect@0.service: Failed with result 'signal'
```

- [x] Journald log 
[journald_v1.99.8-79-g246c6f9_os_reboot.txt](https://github.com/openbmc/openbmc/files/1174093/journald_v1.99.8-79-g246c6f9_os_reboot.txt)

geissonator, 2017-07-25:
The problem is that with the new target refactoring, the host state indicates stopped before the chassis target has fully run to off.  This is per design, but doesn't work well with the reboot.  In theory, a reboot via the host should not result in a chassis power cycle, but to keep things simple, that's what we do currently.  I see a few options:
- Move the reboot to the chassis
- Add some logic in the host state code to ensure we complete the chassis off target before a reboot
- Ensure the host target does not complete until the chassis target completes (via After= syntax). 

I'm going to start with option 3 since it's simplest and see if it handles this.  We can revisit when we decide to support a host reboot without a chassis power cycle.

geissonator, 2017-07-25:
Fix tested out well - https://gerrit.openbmc-project.xyz/#/c/5505/

closed 2017-07-25 by williamspatrick

gkeishin, 2017-07-27:
Looks good on latest build today

```
root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x0280d000..0x0280e000 !
[                                                  ] 0%IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.80
        op-build-v1.17-200-gc5f1fc5
        buildroot-2017.05-93-g9b91216
        skiboot-v5.7-rc2
        hostboot-76ac1bc
        linux-4.12.1-openpower1-p1d60f80
        petitboot-v1.5.0-pdd07c9b
        machine-xml-5575e7e
        occ-2f3397d
        hostboot-binaries-c24fccd
        capp-ucode-9c73e9f
        sbe-02021c6-pa23442d
[==================================================] 100%
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-98"
VERSION_ID="v1.99.8-98-g06734f3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-98"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-bd1d1e81e194d143bf6a1681ab35d44f77a9db10 #1 Thu Jul 27 02:33:28 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```