openbmc/openbmc #2028: Shield setting inventory property attributes by external interfaces

gkeishin, 2017-07-25:
The current inventory items attributes can be set by the user via REST.  This, we shouldn’t let people update or modify which is set by the host.

This needs to be protected from being written from external interfaces such as REST or IPMI.

gkeishin, 2017-07-27:
This may not an immediate requirement but needed this at some point of time later before GA
@geissonator  @williamspatrick

anoo1, 2017-09-05:
Per Patrick, this could be specified in the yaml file and have the sbusplus bindings honor the specification. The challenge would be if different implementations want different behavior.

rfrandse, 2018-08-17:
<https://gerrit.openbmc-project.xyz/12019> property: Add support for const (read-only) values
Closes  #2028 Shield setting inventory property attributes by external interfaces

closed 2018-08-21 by geissonator