openbmc/openbmc #2029: intermittent mapper failure - Error starting event loop: No route to host

geissonator, 2017-07-25:
We're seeing this intermittently in CI, happens after a BMC reset.  

```
Jul 25 19:49:51 witherspoon systemd[1]: Created slice system-mapper\x2dwait.slice.
Jul 25 19:50:11 witherspoon kernel[979]: [   14.180000] systemd[1]: Created slice system-mapper\x2dwait.slice.
Jul 25 19:50:23 witherspoon phosphor-hwmon-readd[952]: Error in mapper call
Jul 25 19:50:27 witherspoon phosphor-hwmon-readd[1006]: Error in mapper call
Jul 25 19:50:31 witherspoon phosphor-hwmon-readd[1008]: Error in mapper call
Jul 25 19:50:34 witherspoon phosphor-hwmon-readd[1011]: Error in mapper call
Jul 25 19:50:38 witherspoon phosphor-hwmon-readd[1014]: Error in mapper call
Jul 25 19:50:40 witherspoon phosphor-mapper[973]: ObjectMapper startup complete.  Discovery in progress...
Jul 25 19:50:41 witherspoon phosphor-hwmon-readd[1019]: Error in mapper call
Jul 25 19:50:41 witherspoon phosphor-mapper[973]: Traceback (most recent call last):
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/sbin/phosphor-mapper", line 45, in <module>
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     interface_blacklists=args.interface_blacklists.split()))
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 773, in server_main
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     interface_blacklists)
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 273, in __init__
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     self.discover()
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 483, in discover
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     owners = [self.bus.get_name_owner(x) for x in owned_names]
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 361, in get_name_owner
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     's', (bus_name,), **keywords)
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Jul 25 19:50:41 witherspoon phosphor-mapper[973]:     message, timeout)
Jul 25 19:50:41 witherspoon phosphor-mapper[973]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'xyz.openbmc_project.Hwmon.hwmon9': no such name
Jul 25 19:50:42 witherspoon mapper[1029]: Error starting event loop: No route to host
Jul 25 19:50:42 witherspoon mapper[1027]: Error starting event loop: No route to host
Jul 25 19:50:42 witherspoon mapper[1028]: Error starting event loop: No route to host
Jul 25 19:50:43 witherspoon mapper[1041]: Error starting event loop: No route to host
Jul 25 19:50:43 witherspoon mapper[1040]: Error starting event loop: No route to host
Jul 25 19:50:43 witherspoon mapper[1039]: Error starting event loop: No route to host
Jul 25 19:50:43 witherspoon mapper[1037]: Error starting event loop: No route to host
Jul 25 19:50:43 witherspoon mapper[1034]: Error starting event loop: No route to host
```

Based on latest code, unrelated to the change being tested (have seen on multiple other gerrit commits).  A reboot of the BMC seems to fix it.
```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-80"
VERSION_ID="v1.99.8-80-gfd9ca7e"
```

It's a bad one because everything starts to fail and we never get to BMC standby.

geissonator, 2017-07-25:
If it's the hwmon9 causing the issue, here's where it maps to:
```
[witherspoon]# ls -al /sys/class/hwmon/hwmon9/
total 0
drwxr-xr-x    3 root     root             0 Jul 25 19:49 .
drwxr-xr-x    3 root     root             0 Jul 25 19:49 ..
lrwxrwxrwx    1 root     root             0 Jul 25 19:50 device -> ../../../iio-hwmon-dps310
-r--r--r--    1 root     root          4096 Jul 25 20:22 name
lrwxrwxrwx    1 root     root             0 Jul 25 19:50 of_node -> ../../../../../firmware/devicetree/base/iio-hwmon-dps310
drwxr-xr-x    2 root     root             0 Jul 25 20:22 power
lrwxrwxrwx    1 root     root             0 Jul 25 19:49 subsystem -> ../../../../../class/hwmon
-r--r--r--    1 root     root          4096 Jul 25 19:50 temp1_input
-rw-r--r--    1 root     root          4096 Jul 25 19:49 uevent
```

spinler, 2017-07-25:
that dps310 definitely will crash hwmon still.

geissonator, 2017-07-25:
Yeah, guess I'm not sure if it's related to the mapper error here or not

geissonator, 2017-07-25:
Full Journal:
[journal.txt](https://github.com/openbmc/openbmc/files/1174721/journal.txt)

geissonator, 2017-07-26:
Looks like the dps310 pops up very early in the boot, that causes phosphor-hwmon to start and interfere with mapper startup.  Probably a bug in mapper that this happens but for now, I put up a fix to ensure mapper is running before the phosphor hwmon app starts.
https://gerrit.openbmc-project.xyz/#/c/5560/

closed 2017-07-27 by williamspatrick