openbmc/openbmc #2031: BMC code update: activation flow review issues

mtritz, 2017-07-25:
A few small issues were raised in code review for https://github.com/openbmc/openbmc/issues/1756. These should be addressed as a prerequisite to https://github.com/openbmc/openbmc/issues/2030 in order to avoid a BMC version of issue https://github.com/openbmc/openbmc/issues/1984.

```

activation.cpp
Line 16:
I would recommend adding these to the configure.ac file
Line 79:
Set booleans back to false, as in https://gerrit.openbmc-project.xyz/#/c/5493/

activation.hpp
Line 130:
Need to unsubscribe at some point?

rfrandse, 2017-07-26:
<https://gerrit.openbmc-project.xyz/5507> Miscellaneous improvements to activation/item updater
Resolves  #2031 BMC code update: activation flow review issues

mtritz, 2017-07-26:
# **STATUS**
7/26: I posted a commit to Gerrit for review that addresses these issues.

closed 2017-08-01 by williamspatrick