openbmc/openbmc #2032: Host reboot followed by hard power off will result in a host power on

geissonator, 2017-07-26:
The host state code monitors for changes to the host state targets.  If a host reboot is requested, it will first call it's host-stop target, when the host state application detects the host-stop target has completed, it will check if the requested action was a reboot, and if so, it will start it's host-start target.  All is good...for now.  

Now someone comes along and does a chassis-hard-poweroff to the system.  The requested host state is still at reboot so the host state application sees it's host-stop target complete, sees it has a reboot request, and powers the system back on...

Patrick and I have never really like putting this reboot logic in the host state application and now I have a good reason to fix it.  I'd like to encapsulate the whole host reboot thing into it's own target.  Something that requires the chassis/host power off's and then as it's last service, it starts the host-start target.  Not sure if possible but I'm gonna give it a shot.  Other ideas welcome!

geissonator, 2017-07-26:
# **STATUS**
7/26 - Coded up initial target and dependencies, need to do the service next and test it all out

rfrandse, 2017-07-27:
<https://gerrit.openbmc-project.xyz/5591> Use new host reboot target to reboot host
Resolves  #2032 Host reboot followed by hard power off will result in a host power on

closed 2017-08-12 by williamspatrick