openbmc/openbmc #2033: OpenBMC incorrectly sends blank network override to host

sammj, 2017-07-26:
In recent OpenBMC versions (at least v1.99.8-71+) Petitboot is receiving an all-zero network override response. Because the response has a zero return code and the valid flags are fine, Petitboot treats it as a real override:

```
IPMI net override resp [0][29]:
 00 01 61 80 21 70 62 21  00 01 06 04 00 00 00 00
 00 00 00 01 00 00 00 00  00 00 00 00 00
Applying IPMI network interface override
configuration:
 autoboot: disabled
 network configuration:
  interface 00:00:00:00:00:00
   static:
    ip:  0.0.0.0/0
    gw:  0.0.0.0
    url:  (null)
  boot device 0: network
  boot device 1: any
  IPMI boot device 0x00
  Modifications allowed to disks: yes
  Default UI to boot on: /dev/hvc0 [IPMI / Serial]
 language: 
```

This will override any existing network config on every boot, replacing it with 0.0.0.0. If there's no network override set the blank default should not be sent to the host - either the return code needs to be properly set or the response truncated.

tomjoseph83, 2017-07-27:
@sammj Can you summarise which IPMI command is behaving differently than earlier?

sammj, 2017-07-28:
This is the network override extension of the "Get Boot Options" IPMI command. In OpenBMC I believe this is handled here: https://github.com/openbmc/phosphor-host-ipmid/blob/master/chassishandler.cpp#L1281

Since this breaks network this regression breaks network autoboot from Petitboot on every boot.

tomjoseph83, 2017-07-31:
We need to handle the case where the if the IP address is set to the default value, then we need to send a IPMI error completion code.

dkodihal, 2017-08-04:
Apart from the fix @ratagupt is going to push, need these settings fixes as well :

https://gerrit.openbmc-project.xyz/#/c/5775/
https://gerrit.openbmc-project.xyz/#/c/5776/

rfrandse, 2017-08-04:
<https://gerrit.openbmc-project.xyz/5773> Don't send blank network override to host
Resolves  #2033 OpenBMC incorrectly sends blank network override to host

ratagupt, 2017-08-04:
Hopefully with these three fixes 2087 also goes.
I am waiting for the submitter response.

closed 2017-08-08 by williamspatrick

ralphbellofatto, 2017-08-15:
as of this set of firmware:
```
IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.87
	op-build-v1.17-210-gbabb528
	buildroot-2017.05-93-g9b91216
	skiboot-v5.7
	hostboot-f523eaa
	linux-4.12.1-openpower1-p8927fdf
	petitboot-v1.5.1-pce8abef
	machine-xml-5575e7e
	occ-ca84830
	hostboot-binaries-c24fccd
	capp-ucode-9c73e9f
	sbe-02021c6-p6a0f00f
```

The problem is still with us.

stewart-ibm, 2017-08-15:
@rahulmah that's host firmware, this bug is on the BMC side. the host firmware has nothing to do with it.

ralphbellofatto, 2017-08-15:
the bmc version shows:
```
BMC Firmware Product: ibm-v1.99.8-98-g6734f31-r2-0-g752ced3 (Active)
```

stewart-ibm, 2017-08-15:
my guess is @williamspatrick can work out if that build has the fix  for this issue or not.

sammj, 2017-08-17:
If I've pieced together everything correctly it looks like the override fix will be in OpenBMC versions `v1.99.8-200-g201bd626` and up - so you'll need to update.

ratagupt, 2017-08-17:
root@witherspoon:/# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.8"
VERSION_ID="ibm-v1.99.8-231-gdf827b69-r0-0-gdf827b6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.8"
BUILD_ID="ibm-v1.99.8-231-gdf827b69-r0"

root@witherspoon:/# /tmp/ipmitool -I dbus raw 0x00 0x09 0x61 0x80 0x00
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x9 rsp=0xff): Unspecified error.

ralphbellofatto, 2017-08-17:
How is this software actually released.  I see a release/tag dated July 12, which appears to be the software that we have released for the witherspoon as 1730c.  ibm-v1.99.8-98-g6734f31-r2-0-g752ced3,

But i don't see any software available as 1.99.8-200-g201bd626 anywhere.  

How do we get this software and how do we update it.?

sammj, 2017-08-18:
@ratagupt I'm not sure what your comment means - are you saying that the IPMI command doesn't work on that version of OpenBMC?

@ralphbellofatto Indeed looks like the OpenBMC image hasn't been updated for Witherspoon - I'll ask around to see what's happening there.

sammj, 2017-08-20:
@ralphbellofatto A new build (with OpenBMC v1.99.8-98-r4) has been released for Witherspoon which should include the fix, please try with that.

ratagupt, 2017-08-22:
@sammj :- What I was mentioning in my last comment that rc is invalid that is the reason for failure of ipmitool command as from the code it returns the Unspecified error.

root@witherspoon:/# /tmp/ipmitool -I dbus raw 0x00 0x09 0x61 0x80 0x00
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x9 rsp=0xff): Unspecified error.
I tested it on the following release.

root@witherspoon:/# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.8"
VERSION_ID="ibm-v1.99.8-231-gdf827b69-r0-0-gdf827b6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.8"
BUILD_ID="ibm-v1.99.8-231-gdf827b69-r0"

sammj, 2017-08-22:
Gotcha, thanks @ratagupt. Pending @ralphbellofatto having a chance to test the fix, this appears to be fixed then.