openbmc/openbmc #2034: [xCAT] How to get time zone information of BMC?

xuweibj, 2017-07-26:
Hi, I am working on getting log information.

```
'/xyz/openbmc_project/logging/entry/40' => {
                 'associations' => [],
                 'Timestamp' => '1501011776382',
                 'Resolved' => 0,
                 'Id' => 40,
                 'Message' => 'org.open_power.Host.Error.Checkstop',
                'Severity' => 'xyz.openbmc_project.Logging.Entry.Level.Error',
                'AdditionalData' => []
               }
```
 I saw that timestamp is ``Commit timestamp of the error event entry in milliseconds since 1970.``  I think we need to get time zone information to know what the exact time of log. 

How could I get it? Or will you provide other time format in the future?

williamspatrick, 2017-07-26:
The BMC only operates in UTC.

xuweibj, 2017-07-27:
OK, Thank you for your comments. I will close this issue.

closed 2017-07-27 by xuweibj