openbmc/openbmc #2036: PNOR image activation fails

charleshofer, 2017-07-26:
Image activation for fails.

```
Jul 25 20:08:49 witherspoon daemon.info systemd[1]: Starting Attach the PNOR mtd device to UBI... Jul 25 20:08:49 witherspoon daemon.info obmc-flash-bios[1198]: ubiattach: error!: cannot attach mtd7 Jul 25 20:08:49 witherspoon daemon.info obmc-flash-bios[1198]:            error 17 (File exists) Jul 25 20:08:50 witherspoon daemon.info obmc-flash-bios[1198]: /usr/sbin/obmc-flash-bios: error 255
```

Steps to replicate:
1. Build an image from https://gerrit.openbmc-project.xyz/#/c/5431/
2. Patch with https://gerrit.openbmc-project.xyz/#/c/5517/, or you can also do `scp /esw/san5/andrewg/mboxd root@<your bmc ip>:/usr/sbin/mboxd`
3. Update PNOR through the REST interface and the activation step should fail

mdmillerii, 2017-07-26:
https://gerrit.openbmc-project.xyz/#/c/5409/3/meta-openbmc-machines/meta-openpower/common/recipes-phosphor/flash/openpower-software-manager/obmc-flash-bios

mdmillerii, 2017-07-26:
... except that is the bmc version and this was the pnor version.

closed 2017-09-26 by charleshofer