openbmc/openbmc #2037: Enable support for usecase specfic bash scipt invocation.

ojayanth, 2017-07-27:
Define generic infrastructure for collecting data.
    Define priority for each type dump  
    Add base scripts used for all types of dump.
    Install this script.

ojayanth, 2017-09-21:
This design approach is adopted from init.d. 
	
Design Flow:  
	•	The /etc/dreport.conf used to keep all dump specific config and settings 
	•	drepoort is responsible for processing input, compression, copy into permanent location and log management.
	•	dreport generates dump specific temporary configuration files which is required for running  user commands.
	•	dreport utility functions need to provide for logging ( info/important/summary/error).
	•	All installed  Dump collection commands script plugins should store in  /usr/share/dump/bmc-base.
	⁃	Script command  file follows  a simple format and set of consistent actions,  which includes 
	⁃	CONFIG [TYPE ..] [priority]  - TYPE - 0..6 priority 0..99 ( Eg : CONFIG 234 93 — include this in TYPE 2,3,4 with priority 93)
	⁃	Methods
	⁃	execute()   — execute command and save in to temporary file.
              ⁃	summary()   — Update summary log file with execution status.

	•	Create  symbolic links in /usr/share/dump/bmc-xx (use case specific path) during initialization based on the control comments defined in the beginning of the script.
	⁃	File name Eg:   /usr/share/dump/bmc-rc1/b93date    — b ( BMC Dump) , 93 - Sequence .Date -command name.
	⁃	 Sequence number in which the programs should be started. This can be used optimizing dump incase size exceeded. 

	•	Depending on type setting, the  report bash script  will execute the programs from one of the following directories. 
	⁃	1 : User Initiated   /usr/share/bmc-userinitiated
	⁃	2 : Application crash  /usr/share/bmc-appcrash
	•	Dump Manager/command line can  executes /usr/bin/dReport bash script , which includes
	⁃	Initialize 
	⁃	dReport name
	⁃	Create Tmp Directory
	⁃	Create Dreport log/summary files
	⁃	Set the dump Type
	⁃	Initialize dump size
	⁃	Set the type specific parameter ( PID/ Debug File path)
	⁃	Update temporary config file
	⁃	Initialize the summary
	⁃	Update the summary log with available info
	⁃	Collect data
	⁃	Run the scripts specific to the type specific directory
	⁃	size check and continue
	⁃	Generate compressed tarball and save into permanent location.
	⁃	For any general failure which cause Dump collection script generate summary dump will exit.

BMC Dump configuration - /etc/bmcdump.conf. 
	•	Options are set using 'ini'-style name = value pairs. 
	•	Name info
	⁃	filelocationtemp    // Temporary file location in RAM
	⁃	filelocation        // File path in Flash 
	⁃	compression         // Compression scheme.
	⁃	corefile            // Core file path
	⁃	maxsize             // Max file size.
	⁃	debug               // Enable Debug option. to print more trace
======

Dump specific configuration - /<tmp path>/<dump path>/dreport.conf. 

	•	Options are set using 'ini'-style name = value pairs. 
	•	Name info
	⁃	dreport tempfilelocation    // Temporary file location in RAM
	⁃	additional debug data path.
	⁃	Process ID.
======

devenrao, 2017-09-22:
Recipe changes
- [ ] Parse bash scripts for type of the error and priority using either python or bash scripts in recipe 
- [ ] Install the scripts to the corresponding error type folder
- [ ] Copy and create symbol links to the bash scripts
- [ ] Name the file names by appending priority value to the bash script name

ojayanth, 2017-09-22:
This issue will be used foir tracking recipe changes.

rfrandse, 2017-10-17:
<https://gerrit.openbmc-project.xyz/7220> Enable support for usecase specific bash script invocation
Resolves  #2037 Enable support for usecase specfic bash script invocation

closed 2017-10-26 by zahrens