openbmc/openbmc #2039: phosphor-logging should not use mapper on commit interface

geissonator, 2017-07-27:
See #2029  Further discussion on this made us realize that because the commit interface is internal to phosphor-logging, it should not need to use mapper.

Also, how do we ensure applications that load libraries in have their dependencies set up correctly?  This one may need it's own issue.

geissonator, 2017-07-27:
FYI @dkodihal @devenrao

dkodihal, 2017-07-27:
@geissonator the logging server does not use it. The client lib does, because it runs in the client's process.

geissonator, 2017-07-27:
Still, the client lib is tied directly to the server interface, so seems like we could skip mapper.

bradbishop, 2019-09-30:
I still think this would be a good change to make.

spinler, 2020-04-01:
I will take care of this.

spinler, 2020-07-30:
Resolved by https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-logging/+/35212

geissonator, 2021-06-07:
Gerrit commit (https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-logging/+/35212) merged so good to close.

closed 2021-06-07 by geissonator