openbmc/openbmc #2041: Witherspoon water cool jumper reporting incorrectly

msbarth, 2017-07-27:
Verified jumper orientation to workbook spec for pass2 board and air cooled jumper setting is reporting water cooled as well. (per SW395244)

bjwyman, 2017-07-27:
# **STATUS**
07/27:
- Today: @msbarth created this issue based off a CQ defect that had a list of 5 or 6 issues.
  - Issue assigned to me, as I did most of the development work on the application that sets these properties.
  - Attempted to get verification of jumper position from workbook owners. No response.
  - Requested workbook be updated to include pass-1 Sequoia jumper position diagram.
  - Tested on pass-1 Sequoia (some confusion on engineer thinking jumpered for WaterCooled).
  - Tested on pass-2 Sequoia, appears to agree with how we interpret workbook diagram.
  - Tested on Redbud, expected AirCooled only, but with patch, WaterCooled was set to true.
  - Pushed code change to Gerrit for review.
    - https://gerrit.openbmc-project.xyz/#/c/5592/
- Tomorrow: Out.
- Blockers:
  - Need jumper position on Redbud system checked/verified.
  - Need clarification on pass-1 Sequoia planar jumper position from engineer.
  - Need workbook update for pass-1 Sequoia jumper position diagram.
  - Need confirmation that pass-2 Sequoia and Redbud diagram is correct, and interpreted correctly.

rfrandse, 2017-08-01:
<https://gerrit.openbmc-project.xyz/5592> Switch which GPIO value sets air or water cooled
Resolves  #2041 Witherspoon water cool jumper reporting incorrectly

closed 2017-08-02 by williamspatrick