openbmc/openbmc #2042: IPL: Error log cleanup: Pervasive Led.Fru.Monitor.Error.InventoryPathError during power offs on rom123bmc with 910.1730.20170726o

cdeadmin, 2017-07-27:


geissonator, 2017-07-27:
@mine260309 - may want to talk with @vishwabmc or @dkodihal - I think something went in that probably requires some inventory updates on romulus in the LED area.

dhruvibm, 2017-07-28:
There is a callout without an inventory path is this valid?
Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=174  ReplyCookie=10
 Sender=:1.6  Destination=:1.45
 UniqueName=:1.6
 MESSAGE "v" {
          VARIANT "a(sss)" {
                  ARRAY "(sss)" {
                          STRUCT "sss" {
                                  STRING "callout";
                                  STRING "fault";
                                  STRING "";
                          };
                  };
          };
 };

mine260309, 2017-08-03:
Most likely, this issue can be fixed with https://gerrit.openbmc-project.xyz/#/c/5152/5 merged.

gtmills, 2018-08-14:
@mine260309 Can we close?

mine260309, 2018-08-15:
Yes, close this please.