openbmc/openbmc #2044: [Palmetto] Inventory Path Error

swe12345, 2017-07-28:
Driver Info:
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-104"
VERSION_ID="v1.99.8-104-gd108c56"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-104"
BUILD_ID="v1.99.8"
```
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```
```
  "Id": 43,
      "Message": "org.open_power.Host.Event.Error.Event",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501219136881,
      "associations": [
        [
          "callout",
          "fault",
          ""
        ]
      ]
    },
    "/xyz/openbmc_project/logging/entry/43/callout": {
      "endpoints": [
        ""
      ]
    },
    "/xyz/openbmc_project/logging/entry/44": {
      "AdditionalData": [],
      "Id": 44,
      "Message": "xyz.openbmc_project.Led.Fru.Monitor.Error.InventoryPathError",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501219137248,
      "associations": []
    },
    "/xyz/openbmc_project/logging/entry/45": {
      "AdditionalData": [
        "CALLOUT_INVENTORY_PATH=/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core5",
        "ESEL=00 00 df 00 00 00 00 20 00 04 fe 21 07 aa 00 00 50 48 00 30 01 00 09 00 00 00 00 01 78 20 80 05 00 00 00 01 d8 25 d5 9c 42 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 90 00 03 ee 90 00 03 ee 55 48 00 18 01 00 09 00 23 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 09 0f 00 48 00 00 00 e0 00 00 02 00 00 00 00 00 02 20 00 00 00 00 00 00 00 20 dc 1b 00 00 00 00 00 00 00 00 42 43 32 33 30 39 30 46 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 02 00 00 00 00 00 00 00 0e 23 01 00 02 00 03 03 00 55 44 00 0c 01 01 09 00 00 20 dc 1b 55 44 00 10 01 02 09 00 3a 62 c8 1d 00 00 00 00 55 44 00 48 01 03 01 00 00 00 00 00 00 03 f3 e4 00 00 00 00 00 03 b6 74 00 00 00 00 40 0a 97 70 00 00 00 00 40 0a b4 c8 00 00 00 00 40 14 e7 8c 00 00 00 00 40 14 85 58 00 00 00 00 40 14 ca 64 00 00 00 00 00 00 25 64 55 44 00 1c 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 00 55 44 00 38 01 04 01 00 07 5f 1d f4 53 4c 34 44 33 31 36 52 31 31 53 38 4b 46 20 20 20 20 00 00 0b ac 54 02 42 32 55 56 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 44 01 00 01 15 31 00 01 28 00 42 46 41 50 49 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f8 00 00 00 00 00 00 00 f8 00 00 00 06 00 00 00 00 00 00 00 0c 0c 16 56 92 00 00 01 9c 00 20 43 4f 02 27 7e 18 00 00 00 62 63 65 6e 74 61 75 72 20 20 20 20 6b 30 3a 6e 30 3a 73 30 3a 70 30 31 20 20 20 20 20 20 20 00 00 00 00 00 3c 00 00 00 0c 11 b1 e5 f6 00 00 01 9d 00 08 43 4f 46 9c 0c 8d 00 00 01 05 00 00 00 00 00 00 00 00 00 00 00 24 00 00 00 0c 12 40 21 e3 00 00 01 9d 00 00 43 4f 0c e7 38 16 00 00 00 74 00 00 00 1c 00 00 00 0c 13 5c a4 a6 00 00 01 9d 00 00 43 4f 0c e7 38 16 00 00 00 74 00 00 00 1c 00 00 00 0c 13 5d 57 77 00 00 01 9d 00 00 43 4f 0c e7 38 16 00 00 00 74 00 00 00 1c 00 00 00 0c 13 5d f8 de 00 00 01 9d 00 00 43 4f 0c e7 38 16 00 00 00 74 00 00 00 1c 55 44 01 80 01 15 31 00 01 28 00 42 46 41 50 49 5f 49 00 00 00 00 00 00 00 00 00 00 00 00 01 78 00 00 00 00 00 00 01 78 00 00 00 0a 00 00 00 00 00 00 00 0c 12 40 25 e4 00 00 01 9d 00 08 43 4f d7 d5 2a d5 00 00 02 bc 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5c 97 a4 00 00 01 9d 00 00 43 4f e6 f2 27 85 00 00 01 26 00 00 00 1c 00 00 00 0c 13 5c 9b 78 00 00 01 9d 00 08 43 4f 47 dd 74 14 00 00 00 c4 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5c a8 94 00 00 01 9d 00 08 43 4f d7 d5 2a d5 00 00 02 bc 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5d 49 87 00 00 01 9d 00 00 43 4f e6 f2 27 85 00 00 01 26 00 00 00 1c 00 00 00 0c 13 5d 4e 1b 00 00 01 9d 00 08 43 4f 47 dd 74 14 00 00 00 c4 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5d 5b 2b 00 00 01 9d 00 08 43 4f d7 d5 2a d5 00 00 02 bc 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5d ed 5c 00 00 01 9d 00 00 43 4f e6 f2 27 85 00 00 01 26 00 00 00 1c 00 00 00 0c 13 5d f1 1d 00 00 01 9d 00 08 43 4f 47 dd 74 14 00 00 00 c4 00 00 00 00 00 20 dc 1b 00 00 00 24 00 00 00 0c 13 5d fc ac 00 00 01 9d 00 08 43 4f d7 d5 2a d5 00 00 02 bc 00 00 00 00 00 20 dc 1b 00 00 00 24 55 44 00 30 01 15 31 00 01 28 00 42 53 43 41 4e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 28 00 00 00 00 00 00 00 28 00 00 00 00 00 00 00 00"
```
```

Jul 28 05:18:52 palmetto ipmid[963]: SUCCESS: Failed to get /system/chassis/motherboard/dimm0 busname: (null)
Jul 28 05:18:52 palmetto ipmid[963]: A host system event was received
Jul 28 05:18:52 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/40
Jul 28 05:18:52 palmetto phosphor-fru-fault-monitor[889]: Invalid Inventory Path.
Jul 28 05:18:52 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/41
Jul 28 05:18:54 palmetto ipmid[963]: Failed to get /system/chassis/motherboard/dimm1 busname: (null)
Jul 28 05:18:54 palmetto ipmid[963]: A host system event was received
Jul 28 05:18:54 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/42
Jul 28 05:18:56 palmetto ipmid[963]: Failed to get /system/chassis/motherboard/dimm2 busname: (null)
Jul 28 05:18:56 palmetto ipmid[963]: A host system event was received
Jul 28 05:18:56 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/43
Jul 28 05:18:57 palmetto phosphor-fru-fault-monitor[889]: Invalid Inventory Path.
Jul 28 05:18:57 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/44
Jul 28 05:18:59 palmetto ipmid[963]: Failed to get /system/chassis/motherboard/dimm3 busname: (null)
Jul 28 05:18:59 palmetto ipmid[963]: A host system event was received
Jul 28 05:18:59 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/45
Jul 28 05:19:01 palmetto system_manager.py[870]: ERROR SystemManager: 53 not found in lookup
Jul 28 05:19:01 palmetto ipmid[963]: A host system event was received
Jul 28 05:19:01 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/46
Jul 28 05:19:02 palmetto phosphor-fru-fault-monitor[889]: Invalid Inventory Path.
Jul 28 05:19:02 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/47
Jul 28 05:19:04 palmetto system_manager.py[870]: ERROR SystemManager: 53 not found in lookup
Jul 28 05:19:04 palmetto ipmid[963]: A host system event was received
Jul 28 05:19:04 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/48
Jul 28 05:19:04 palmetto phosphor-fru-fault-monitor[889]: Invalid Inventory Path.
Jul 28 05:19:05 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/49
Jul 28 05:19:07 palmetto system_manager.py[870]: ERROR SystemManager: 53 not found in lookup
Jul 28 05:19:07 palmetto ipmid[963]: A host system event was received
Jul 28 05:19:07 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/50
Jul 28 05:19:07 palmetto phosphor-fru-fault-monitor[889]: Invalid Inventory Path.
Jul 28 05:19:07 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/51
Jul 28 05:20:47 palmetto phosphor-watchdog[1203]: Timer Expired
Jul 28 05:20:47 palmetto systemd[1]: Starting OpenPOWER debug data collector for host watchdog timeout...
Jul 28 05:20:47 palmetto watchdog_timeout[1237]: Host watchdog timed out
Jul 28 05:20:48 palmetto phosphor-fru-fault-monitor[889]: /xyz/openbmc_project/logging/entry/52
Jul 28 05:20:48 palmetto systemd[1]: Started OpenPOWER debug data collector for host watchdog timeout.
Jul 28 05:20:48 palmetto systemd[1]: Reached target Quiesce Target.
Jul 28 05:20:48 palmetto systemd[1]: Reached target Host instance 0 timed out refreshing Watchdog.
Jul 28 05:20:48 palmetto phosphor-host-state-manager[919]: Auto reboot enabled. HOST BOOTCOUNT already set to 0.
Jul 28 05:20:48 palmetto sensor_manager2.py[813]: Setting boot count to 3
Jul 28 05:20:48 palmetto phosphor-host-state-manager[919]: Maintaining quiesce
```
[journalctl_28jul.txt](https://github.com/openbmc/openbmc/files/1182118/journalctl_28jul.txt)

swe12345, 2017-07-28:
Observing the same  in Barreleye as well
Driver Info:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-104"
VERSION_ID="v1.99.8-104-gd108c56"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-104"
BUILD_ID="v1.99.8"
```
```
"Id": 8,
      "Message": "org.open_power.Host.Event.Error.Event",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501015772705,
      "associations": [
        [
          "callout",
          "fault",
          ""
        ]
      ]
    },
    "/xyz/openbmc_project/logging/entry/8/callout": {
      "endpoints": [
        ""
      ]
    },
    "/xyz/openbmc_project/logging/entry/9": {
      "AdditionalData": [],
      "Id": 9,
      "Message": "xyz.openbmc_project.Led.Fru.Monitor.Error.InventoryPathError",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501015773098,
      "associations": []
    }
  },
  "message": "200 OK",
  "status": "ok"
```
[journalctl_28_barreleye.txt](https://github.com/openbmc/openbmc/files/1182175/journalctl_28_barreleye.txt)

gtmills, 2018-07-05:
@swe12345 This is an old build.. Still an issue?

gkeishin, 2018-08-20:
Probably we can close the palmetto ones. And re-open when we retest them?

gtmills, 2018-08-20:
@gkeishin At least the ones that are probably no longer valid. This is a very old driver and hasn't been updated in 13 months.  Thanks!

closed 2018-08-20 by gkeishin