openbmc/openbmc #2045: max31785 3-0052: Fan 0 is configured but not installed

gkeishin, 2017-07-28:
We hit this on multiple systems and it failed to boot host. On debugging with @amboar we found the FAN 

```
root@witherspoon:~# dmesg | grep -i fan
[    4.400000] max31785 3-0052: Fan 0 is configured but not installed
[    4.410000] max31785 3-0052: Fan 1 is configured but not installed
[    4.420000] max31785 3-0052: Fan 2 is configured but not installed
[    4.430000] max31785 3-0052: Fan 3 is configured but not installed
root@witherspoon:~#
```

- [x] dmesg 
[dmesg_kernel.txt](https://github.com/openbmc/openbmc/files/1182154/dmesg_kernel.txt)

`Driver info`

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-104"
VERSION_ID="v1.99.8-104-gd108c56"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-104"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-44f0dc8ac43162c4f7082884d6cd79400341bf5b #1 Fri Jul 28 03:21:13 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```

sivassrr, 2017-07-28:
This is causing system not to boot, right.

amboar, 2017-07-28:
Okay, this comes back to an assumption that I probably should have dug a little deeper on. For all the systems I tested with, bit 7 of `FAN_CONFIG_1_2` was set in hardware after a PDU cycle. Bit 7 is the "Fan Installed" bit, and I used this as a sanity check against the system's devicetree description. I assumed this was configured in the default or user memory of the MAX31785 to reflect the system hardware, or was set by the device itself (in some manner).

If that's not the case and the kernel must instead set the bit then that's fine, I just need to know what the *intended* or *designed* behaviour is. It seems like this is a policy thing regardless, and it might be best if I consider this in the devicetree bindings.

amboar, 2017-07-28:
@sivassrr this issue prevents the host from booting, as phosphor-hwmon can't find the fans it requires.

gkeishin, 2017-08-10:
https://github.com/openbmc/openbmc/issues/2047

amboar, 2017-08-10:
Are we still seeing this? In my testing it was resolved by https://github.com/openbmc/linux/commit/03afe5775baf06c3037d62cc2a15628079af38c9

gkeishin, 2017-08-10:
I don't think I see it in my run anymore.  @amboar

amboar, 2017-08-10:
I think we can close it.

closed 2017-08-10 by gkeishin