openbmc/openbmc #2050: PNOR code update: Extend factory reset

mtritz, 2017-07-30:
Since the original implementation of factory reset, the footprint of the host updater has increased, and the factory reset isn't getting everything cleaned up. Some things that ought to be added to fully reset the system:

- Unmount and remove read-only volumes created by activation
- clear symlinks to priority 0 activation
- erase activation dbus objects
- erase version dbus objects
- remove the serial files that persist settings (currently just redundancy priority) on reset
- clear /tmp/images ?

mtritz, 2017-07-30:
- [ ] Unmount and remove read-only volumes created by activation
- [ ] Erase activation dbus objects
- [ ] Erase version dbus objects
- [ ] Remove the serial files that persist settings on reset

mtritz, 2017-07-30:
# **STATUS**
7/30: Placed a commit in Gerrit that follows the checklist above. Need design discussion: should symlinks be removed? Should /tmp/images be cleared?

rfrandse, 2017-07-30:
<https://gerrit.openbmc-project.xyz/5644> Host updater: Factory reset extension
Resolves  #2050 PNOR code update: Extend factory reset

mtritz, 2017-07-31:
Per design meeting, factory reset is good as-is.

closed 2017-07-31 by mtritz