openbmc/openbmc #2052: Host boot fails after two PNOR updates and a switch back to the first image

charleshofer, 2017-07-30:
1. Do a PNOR update
2. Start the host and shut it down after boot was successful
3. Do a PNOR update with a different image
4. Start the host and shut it down after boot was successful
5. Switch back to the original PNOR image by changing the Priority
6. Attempt to start the host again, and this fails

[journal.txt](https://github.com/openbmc/openbmc/files/1185897/journal.txt)

geissonator, 2017-07-30:
We've got a custom image from @saqibkh based on master that picked up all code update fixes as well as the fixes from @amboar for the max chip.

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-124"
VERSION_ID="v1.99.8-124-g06e07ed"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-124"
BUILD_ID="v1.99.8"
[witherspoon]# uname -a
Linux witherspoon 4.10.17-44f0dc8ac43162c4f7082884d6cd79400341bf5b #1 Sun Jul 30 09:46:37 CDT 2017 armv6l GNU/Linux
```
```
vJul 30 15:05:11 witherspoon systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Jul 30 15:05:11 witherspoon phosphor-fan-presence-tach[3531]: Failed to invoke DBus method.
Jul 30 15:05:11 witherspoon phosphor-fan-presence-tach[3531]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 30 15:05:11 witherspoon phosphor-fan-presence-tach[3531]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Jul 30 15:05:11 witherspoon phosphor-fan-presence-tach[3531]: The operation failed internally.
Jul 30 15:05:11 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Main process exited, code=dumped, status=6/ABRT
Jul 30 15:05:11 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Unit entered failed state.
Jul 30 15:05:11 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Failed with result 'core-dump'.
Jul 30 15:05:11 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Service hold-off time over, scheduling restart.
Jul 30 15:05:11 witherspoon systemd[1]: Stopped Phosphor Fan Presence Tach Daemon.
Jul 30 15:05:11 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Start request repeated too quickly.
Jul 30 15:05:11 witherspoon systemd[1]: Failed to start Phosphor Fan Presence Tach Daemon.
Jul 30 15:05:11 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Jul 30 15:05:11 witherspoon systemd[1]: Dependency failed for Start Host0.
Jul 30 15:05:11 witherspoon systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Jul 30 15:05:11 witherspoon systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Jul 30 15:05:11 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Jul 30 15:05:11 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
```

spinler, 2017-07-31:
A pile of things are in the works to make this go away:
1) MAX31785 device driver is being rewritten so it actually works as it should
2) fan presence will stop terminating on missing dbus properties
3) fan presence is being moved out of the poweron path and will start at standby instead
4) fan presence will look at GPIOs first, which bypasses dbus/hwon/device driver

geissonator, 2017-08-04:
We've been good since picking up the new max chip driver.

closed 2017-08-04 by geissonator