openbmc/openbmc #2060: Implement minimum ship level

bradbishop, 2017-08-01:
Provide a framework for minimum ship level enforcement.
Implement a solution to prevent downlevel fan cards on Witherspoon within the framework.

bradbishop, 2017-08-01:
**TASKLIST**
- [x] Propose and implement framework dbus interfaces.
- [x] Add a setting for enable/disable.
- [] Write an application that sets MeetsMinimumShipLevel on fan inventory on witherspoon
- [] Write an application that validates MeetsMinimumShipLevel against the setting
- [] Insert the validation application into the power on sequence.

# **STATUS**
7/31 - Adding enable/disable setting.  Today plan to write the other two applications.

rfrandse, 2017-08-23:
<https://gerrit.openbmc-project.xyz/6154> wspoon: Run MSL verify service
Resolves:  #2060 Implement minimum ship level

closed 2017-08-24 by bradbishop