openbmc/openbmc #2065: Undefined symbol when loading IPMI provider library

gkeishin, 2017-08-02:
`ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx sel info`
```
Get Device ID command failed: 0xc1 Invalid command
Get SEL Info command failed: Invalid command
```

```
Jul 21 11:34:55 witherspoon netipmid[1757]: >> closeSession
Jul 21 11:34:55 witherspoon netipmid[1757]: << closeSession
Jul 21 11:34:55 witherspoon netipmid[1757]: >> GetChannelCapabilities
Jul 21 11:34:55 witherspoon netipmid[1757]: << GetChannelCapabilities
Jul 21 11:34:55 witherspoon netipmid[1757]: >> openSession
Jul 21 11:34:55 witherspoon netipmid[1757]: << openSession
Jul 21 11:34:55 witherspoon netipmid[1757]: >> RAKP12
Jul 21 11:34:55 witherspoon netipmid[1757]: << RAKP12
Jul 21 11:34:55 witherspoon netipmid[1757]: >> RAKP34
Jul 21 11:34:55 witherspoon netipmid[1757]: << RAKP34
Jul 21 11:34:55 witherspoon netipmid[1757]: >> setSessionPrivilegeLevel
Jul 21 11:34:55 witherspoon netipmid[1757]: << setSessionPrivilegeLevel
Jul 21 11:34:55 witherspoon netipmid[1757]: E> Table:: Command Not found: 0x1801
Jul 21 11:34:55 witherspoon netipmid[1757]: E> Table:: Command Not found: 0xb03e
Jul 21 11:34:55 witherspoon netipmid[1757]: E> Table:: Command Not found: 0xb000
Jul 21 11:34:55 witherspoon netipmid[1757]: E> Table:: Command Not found: 0x2840
```

So we restarted the IPMI service and see this
```
Jul 21 11:38:40 witherspoon systemd[1]: Stopping Network IPMI daemon...
Jul 21 11:38:40 witherspoon systemd[1]: Stopped Network IPMI daemon.
Jul 21 11:38:40 witherspoon systemd[1]: Started Network IPMI daemon.
Jul 21 11:38:40 witherspoon netipmid[2288]: Registering handler: /usr/lib/net-ipmid/libapphandler.so.0
Jul 21 11:38:40 witherspoon netipmid[2288]: Error opening /usr/lib/net-ipmid/libapphandler.so.0/usr/lib/net-ipmid/libapphandler.so.0: undefined symbol: _ZN8settings7ObjectsC1ERN9sdbusplus3bus3busERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISB_EE
```
`Driver info`
This is on latest Master build
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-147"
VERSION_ID="v1.99.8-147-g9a33a15"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-147"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-03afe5775baf06c3037d62cc2a15628079af38c9 #1 Tue Aug 1 19:23:53 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

dkodihal, 2017-08-02:
Need to pull in this commit into net-ipmid as well - https://gerrit.openbmc-project.xyz/#/c/5477/.

rfrandse, 2017-08-02:
<https://gerrit.openbmc-project.xyz/5735> netipmid: Provide support to fetch paths of settings dbus objects
Resolves  #2065 Undefined symbol when loading IPMI provider library

tomjoseph83, 2017-08-02:
https://gerrit.openbmc-project.xyz/#/c/5735/

tomjoseph83, 2017-08-07:
# **STATUS**

8/7
Ready for merge review.

closed 2017-08-08 by williamspatrick