openbmc/openbmc #2066: ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22

PavamanSubramaniyam, 2017-08-02:
I have upgraded the Open BMC Firmware image on Witherspoon P9 Hardware to the latest master build.
https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/930/artifact/deploy/images/witherspoon/witherspoon-20170801191648.all.tar

In the dmesg I keep on seeing the below messages displayed.

```
[   24.820000] systemd-journald[547]: Received request to flush runtime journal from PID 1
[   37.110000] ubi7: default fastmap pool size: 100
[   37.110000] ubi7: default fastmap WL pool size: 50
[   37.110000] ubi7: attaching mtd7
[   37.580000] ubi7: scanning is finished
[   37.580000] ubi7 error: ubi_read_volume_table: the layout volume was not found
[   37.580000] ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
[   48.330000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
```

The machine is installed with the following level of BMC and PNOR Firmware images.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-147"
VERSION_ID="v1.99.8-147-g9a33a15"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-147"
BUILD_ID="v1.99.8"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x0280d000..0x0280e000 !
[                                                  ] 0%open-power-witherspoon-v1.18-14-g6ab8d5a
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-24-g81c0e076374e-pc665224
        hostboot-74fe231
        linux-4.12.3-openpower1-p3f3388a
        petitboot-v1.5.1-pb361a08
        machine-xml-5575e7e
        occ-458a999
        hostboot-binaries-c24fccd
        capp-ucode-9c73e9f
        sbe-02021c6-p757955a
[==================================================] 100%
```

gkeishin, 2017-08-02:
oops we opened at the same time :) 

https://github.com/openbmc/openbmc/issues/2067

I close my defect up and track with this one

saqibkh, 2017-08-02:
This is just an additional log which I think we can prevent by checking the pnormtd device before trying to attach the device to ubi.

This log should automatically go away once we move to the new PNOR Code Update that uses UBI. But for all other systems that don't use UBI we will need to fix this error log.

rfrandse, 2017-08-07:
<https://gerrit.openbmc-project.xyz/5825> PNOR: Check if pnor mtd already attached on reboot.
Resolves  #2066 ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22

closed 2017-08-14 by williamspatrick