openbmc/openbmc #2068: eSEL  Error in mapper call

gkeishin, 2017-08-02:
To recreate this
- [x] Boot to OS
- [x] Issue the following command to send down error event to BMC

```
root@ws013p1:~# ipmitool -C 3 raw 0x0a 0x42
 0d 00
root@ws013p1:~#

ipmitool -C 3 raw 0x32 0xf0 0x0d 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0xdf 0x00 0x00 0x00 0x00 0x20 0x00 0x04 0x12 0x35 0x6f 0xaa 0x00 0x00
 0d 00
root@ws013p1:~#

ipmitool -C 3 raw 0x0a 0x44 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x20 0x00 0x04 0x12 0x35 0x6f 0x02 0x00 0x01
 00 01
root@ws013p1:~#

Jul 21 11:57:38 witherspoon ipmid[1067]: A host system event was received
Jul 21 11:57:38 witherspoon ipmid[1067]: Error in mapper call

$ ./rest_cmd -i xx.xx.xx.xx-o esel
Fetching esel from BMC
No Error entry(s) found:
```

tomjoseph83, 2017-08-17:
Sensor Number 0xA6 would callout /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm0.

1) ipmitool -C 3 raw 0x0a 0x42

2) ipmitool -C 3 raw 0x32 0xf0 0x09 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0xdf 0x00 0x00 0x00 0x00 0x20 0x00 0x04 0x12 0xA6 0x6f 0xaa 0x00 0x00

3)ipmitool -C 3 raw 0x0a 0x44 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x20 0x00 0x04 0x12 0xA6 0x6f 0x02 0x00 0x01

gkeishin, 2017-08-17:
Thanks @tomjoseph83  I fix up in automation with the new commands

closed 2017-08-17 by gkeishin