openbmc/openbmc #2069: BMC failing to respond to IPMI commands intermittently

prkatti1, 2017-08-02:
BMC failing to respond to IPMI commands intermittently 
ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx chassis status
Get Device ID command failed: 0xc1 Invalid command

Version info:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.8"
VERSION_ID="ibm-v1.99.8-98-g06734f31-r2-0-g752ced3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.8"
BUILD_ID="ibm-v1.99.8-98-g06734f31-r2"

PNOR:
IBM-witherspoon-redbud-ibm-OP9_v1.17_1.88
        op-build-v1.17-210-g7c61ec6
        buildroot-2017.05-93-g9b91216
        skiboot-v5.7-paa10293
        hostboot-f523eaa
        linux-4.12.1-openpower1-p1cc6f7f
        petitboot-v1.5.1-p279e9f9
        machine-xml-5575e7e
        occ-ca84830
        hostboot-binaries-c24fccd
        capp-ucode-9c73e9f
        sbe-02021c6-pfad3bc9

Pls find attached detailed logs

prkatti1, 2017-08-02:
[20170802071402086185_IPMIChassisStatusOn.zip](https://github.com/openbmc/openbmc/files/1193652/20170802071402086185_IPMIChassisStatusOn.zip)

gkeishin, 2017-08-02:
I think this is the same https://github.com/openbmc/openbmc/issues/2065

tomjoseph83, 2017-08-02:
@prkatti1 Please close this as it is as dupe of #2065

closed 2017-08-03 by gkeishin