openbmc/openbmc #2072: Make /media tmpfs

anoo1, 2017-08-02:


lgon, 2017-08-11:
Related:  https://gerrit.openbmc-project.xyz/#/c/5705/2/meta-openbmc-machines/meta-openpower/common/recipes-phosphor/flash/openpower-software-manager/obmc-flash-bios

mdmillerii, 2017-08-11:
An alternative to creating an additional `tmpfs` is to make `/media` a symlink to /run/media and create that directory (maybe through systemd `tmpfiles`, or the existing `mkdir -p` in the above reference).

rfrandse, 2017-08-17:
<https://gerrit.openbmc-project.xyz/6037> filesystem: mount rw /media with fstab
Resolves  #2072 Make /media tmpfs

lgon, 2017-08-23:
GitHub Comment
++++++++++++++++++++++++++++++++++++++
TASK LIST

    Version - Add unique pointer to erase callback method from parent.
    Implement Object.Delete in version, and override delete method to call parent's erase method.
    Add erase method to image manager to remove version and activation from maps, and delete tmp image directory. Bind erase method to version when inserted in map.
    Add erase method to item updater to remove version from map and remove ro,rw partitions. Bind erase method to version when inserted in map.
    Create service files to delete read only and read write partitions.

#STATUS Resolved and merged.
8-23 - My change is based off of Edward A. James's changes for https://gerrit.openbmc-project.xyz/#/c/5962/       Once his code is merged, I can make a final test with my changes, and resolve this issue.
++++++++++++++++++++++++++++++++++++++

closed 2017-08-30 by williamspatrick