openbmc/openbmc #2075: Priority of PNOR image decreases to two when setting priority of other image to 0

charleshofer, 2017-08-02:
Given two images A and B on the BMC with priorities 1 and 0 respecitvely, when you set the priority of image A to 0 the priority of image B is set to 1 as it should. However, if you set the priority of image A to 0 again, the priority of image B gets set to 2.
```
[Charles.Hofer@oc4481562561 ~]$ bmc-enumerate 9.3.21.43
{
  "data": {
    "/xyz/openbmc_project/software/193325ab": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "v1.99.8-168-g0bf1180"
    }, 
    "/xyz/openbmc_project/software/2a1022fe": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-176-g11dd356-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2,hostboot-3fa8be9,linux-4.12.1-openpower1-p3a0d8f7,petitboot-v1.5.0-p3eae874,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p1db02ba", 
      "Path": "", 
      "Priority": 1, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    }, 
    "/xyz/openbmc_project/software/40a2d17a": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-131-g75176ac-dirty,buildroot-2017.05-15-gcd06d25,skiboot-v5.7-rc1,hostboot-d0a8f18,linux-4.11.6-openpower1-pc291111,petitboot-v1.4.3-p611c951,machine-xml-site_local-fspbld-bbf60a8,occ-6a00709,hostboot-binaries-0e2ad15,capp-ucode-9c73e9f,sbe-02021c6", 
      "Path": "/tmp/images/40a2d17a", 
      "Priority": 0, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active", 
      "Version": "IBM-witherspoon-ibm-OP9_v1.17_1.58"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}[Charles.Hofer@oc4481562561 ~]$ bmc-put-attr '{"data":0}' 9.3.21.43 software/2a1022fe Priority
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}[Charles.Hofer@oc4481562561 ~]$ bmcenumerate 9.3.21.43
{
  "data": {
    "/xyz/openbmc_project/software/193325ab": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "v1.99.8-168-g0bf1180"
    }, 
    "/xyz/openbmc_project/software/2a1022fe": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-176-g11dd356-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2,hostboot-3fa8be9,linux-4.12.1-openpower1-p3a0d8f7,petitboot-v1.5.0-p3eae874,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p1db02ba", 
      "Path": "", 
      "Priority": 0, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    }, 
    "/xyz/openbmc_project/software/40a2d17a": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-131-g75176ac-dirty,buildroot-2017.05-15-gcd06d25,skiboot-v5.7-rc1,hostboot-d0a8f18,linux-4.11.6-openpower1-pc291111,petitboot-v1.4.3-p611c951,machine-xml-site_local-fspbld-bbf60a8,occ-6a00709,hostboot-binaries-0e2ad15,capp-ucode-9c73e9f,sbe-02021c6", 
      "Path": "/tmp/images/40a2d17a", 
      "Priority": 1, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active", 
      "Version": "IBM-witherspoon-ibm-OP9_v1.17_1.58"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}[Charles.Hofer@oc4481562561 ~]$ bmc-put-attr '{"data":0}' 9.3.21.43 software/2a1022fe Priority
{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}[Charles.Hofer@oc4481562561 ~]$ bmcenumerate 9.3.21.43
{
  "data": {
    "/xyz/openbmc_project/software/193325ab": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "v1.99.8-168-g0bf1180"
    }, 
    "/xyz/openbmc_project/software/2a1022fe": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-176-g11dd356-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2,hostboot-3fa8be9,linux-4.12.1-openpower1-p3a0d8f7,petitboot-v1.5.0-p3eae874,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p1db02ba", 
      "Path": "", 
      "Priority": 0, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    }, 
    "/xyz/openbmc_project/software/40a2d17a": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "ExtendedVersion": "op-build-v1.17-131-g75176ac-dirty,buildroot-2017.05-15-gcd06d25,skiboot-v5.7-rc1,hostboot-d0a8f18,linux-4.11.6-openpower1-pc291111,petitboot-v1.4.3-p611c951,machine-xml-site_local-fspbld-bbf60a8,occ-6a00709,hostboot-binaries-0e2ad15,capp-ucode-9c73e9f,sbe-02021c6", 
      "Path": "/tmp/images/40a2d17a", 
      "Priority": 2, 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active", 
      "Version": "IBM-witherspoon-ibm-OP9_v1.17_1.58"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}[
```

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-168"
VERSION_ID="v1.99.8-168-g0bf1180"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-168"
BUILD_ID="v1.99.8"
```

saqibkh, 2017-08-03:
This is a bug in our code. Before assigning a new priority we check if a version exists with the same priority. If it does then we increment that priority(which in this case is itself) and all subsequent  priorities by 1.

saqibkh, 2017-08-07:
@rfrandse Please assign this defect to me. Thanks

rfrandse, 2017-08-12:
<https://gerrit.openbmc-project.xyz/5959> item_updater: Ignore base version when trying to free up priority.
Resolves  #2075 Priority of PNOR image decreases to two when setting priority of other image to 0

closed 2017-08-16 by williamspatrick