openbmc/openbmc #2078: Enable dump collection support for Internal error

ojayanth, 2017-08-03:
This includes new dump type support in phosphor-dump-collector and some changes in phosphor-logging.

geissonator, 2017-08-10:
Needs proposal from Jayanth and task list

ojayanth, 2017-08-17:
Phosphor-logging “report” function can be used for initiating the “InternalError” type dump.
For any “internal failure” type error commit request, during log report, call the  dump d-bus internal create function with Dump type “InternalFailure” and additional info with errorID.  Dump Manager should trigger the dreport tool to collect the dump.

- [ ] Add InternalError dump type in the Create.interface.yaml.

- [ ] Phosphor-logging: Provide a method to trigger dump during internal failure error commit.
    — Call the  dump d-bus internal interfaces with type Internal error and additional string error id 

- [ ] Dump Manager:  dreport related interface changes if required.

- [ ] dreport:  Add support for “internal error” type reporting.
  — Define type and add input validation.
  — Define contents 
           — Journal log (with meta data)
           — System state and driver info

closed 2017-09-14 by williamspatrick