openbmc/openbmc #2079: Failed to start Create a ubi volume for pnor patches.

gkeishin, 2017-08-03:
`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-175"
VERSION_ID="v1.99.8-175-g6c9f9d8"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-175"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-03afe5775baf06c3037d62cc2a15628079af38c9 #1 Thu Aug 3 02:34:18 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
● obmc-flash-bios-ubipatch.service - Create a ubi volume for pnor patches
   Loaded: loaded (/lib/systemd/system/obmc-flash-bios-ubipatch.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-07-25 19:24:01 UTC; 47s ago
  Process: 978 ExecStart=/usr/sbin/obmc-flash-bios ubimount pnor-patch (code=exited, status=32)
 Main PID: 978 (code=exited, status=32)

Jul 25 19:24:01 witherspoon obmc-flash-bios[978]: libubi: error!: cannot get information about "/dev/ubi7"
Jul 25 19:24:01 witherspoon obmc-flash-bios[978]:         error 2 (No such file or directory)
Jul 25 19:24:01 witherspoon obmc-flash-bios[978]: ubimkvol: error!: error while probing "/dev/ubi7"
Jul 25 19:24:01 witherspoon obmc-flash-bios[978]:           error 2 (No such file or directory)
Jul 25 19:24:01 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Main process exited, code=exited, status=32/n/a
Jul 25 19:24:02 witherspoon obmc-flash-bios[978]: mount: unknown filesystem type 'ubifs'
Jul 25 19:24:01 witherspoon systemd[1]: Failed to start Create a ubi volume for pnor patches.
Jul 25 19:24:02 witherspoon obmc-flash-bios[978]: /usr/sbin/obmc-flash-bios: error 32
Jul 25 19:24:01 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Unit entered failed state.
Jul 25 19:24:01 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Failed with result 'exit-code'
```

- [x] Journald 
[journald_v1.99.8-175-g6c9f9d8_UBI_Error.txt](https://github.com/openbmc/openbmc/files/1195954/journald_v1.99.8-175-g6c9f9d8_UBI_Error.txt)

geissonator, 2017-08-03:
I see one commit came in after this build, not sure if it was needed.  @gkeishin can you describe what you did here?

Update to new BMC, Updated to new PNOR, rebooted BMC - and then saw this error?
Or did you try to apply a patch?

gkeishin, 2017-08-03:
Yeah.. this was with old PNOR.. son on the other trial, we still see this failure with the new PNOR

https://github.com/openbmc/openbmc/issues/2083

gkeishin, 2017-08-03:
1. new PNOR ( squafs)
2.  Update BMC 
3.  Update PNOR pflash
4. reboot

gkeishin, 2017-09-06:
ANother case
```
root@witherspoon:~# system status obmc-flash-bios-ubipatch.service | cat
-sh: system: command not found
root@witherspoon:~# system status obmc-flash-bios-ubipatch.service | cat^C
root@witherspoon:~# systemctl status obmc-flash-bios-ubipatch.service | cat
● obmc-flash-bios-ubipatch.service - Create a ubi volume for pnor patches
   Loaded: loaded (/lib/systemd/system/obmc-flash-bios-ubipatch.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-08-30 18:30:54 UTC; 2min 23s ago
  Process: 1118 ExecStart=/usr/bin/env obmc-flash-bios ubimount pnor-patch (code=exited, status=32)
 Main PID: 1118 (code=exited, status=32)

Aug 30 18:30:53 witherspoon systemd[1]: Starting Create a ubi volume for pnor patches...
Aug 30 18:30:54 witherspoon obmc-flash-bios[1118]: mkdir: can't create directory '/usr/local/': Read-only file system
Aug 30 18:30:54 witherspoon obmc-flash-bios[1118]: mount: mount point /usr/local/share/pnor does not exist
Aug 30 18:30:54 witherspoon obmc-flash-bios[1118]: /usr/sbin/obmc-flash-bios: error 32
Aug 30 18:30:54 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Main process exited, code=exited, status=32/n/a
Aug 30 18:30:54 witherspoon systemd[1]: Failed to start Create a ubi volume for pnor patches.
Aug 30 18:30:54 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Unit entered failed state.
Aug 30 18:30:54 witherspoon systemd[1]: obmc-flash-bios-ubipatch.service: Failed with result 'exit-code'.
root@witherspoon:~#

```

Upload new BMC image, image at Ready state, reset the BMC..

It deletes the non-activated but PNOR services crashed as mentioned above..

- [x] Journald 
[journald_BMC_update_crash.txt](https://github.com/openbmc/openbmc/files/1281323/journald_BMC_update_crash.txt)

mdmillerii, 2017-09-06:
Need ordering and dependency on /usr/local service (which depends on field mode).

anoo1, 2017-09-12:
Check if pnor is in ubi format before attempting to mount ubipatch added here:
https://github.com/openbmc/openbmc/commit/eebcd5fe997f936dd5800ac5eb364e8495e6b9cb

closed 2017-09-12 by anoo1