openbmc/openbmc #2080: mboxd.service: Failed with result 'core-dump'

gkeishin, 2017-08-03:
`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-175"
VERSION_ID="v1.99.8-175-g6c9f9d8"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-175"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-03afe5775baf06c3037d62cc2a15628079af38c9 #1 Thu Aug 3 02:34:18 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
● mboxd.service - Phosphor MBOX Daemon
   Loaded: loaded (/lib/systemd/system/mboxd.service; enabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Tue 2017-07-25 19:25:05 UTC; 5min ago
  Process: 1166 ExecStart=/usr/sbin/mboxd --flash 64M --window-size 1M (code=dumped, signal=ABRT)
 Main PID: 1166 (code=dumped, signal=ABRT)

Jul 25 19:25:04 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
Jul 25 19:25:05 witherspoon systemd[1]: mboxd.service: Service hold-off time over, scheduling restart.
Jul 25 19:25:05 witherspoon systemd[1]: Stopped Phosphor MBOX Daemon.
Jul 25 19:25:05 witherspoon systemd[1]: mboxd.service: Start request repeated too quickly.
Jul 25 19:25:05 witherspoon systemd[1]: Failed to start Phosphor MBOX Daemon.
Jul 25 19:25:05 witherspoon systemd[1]: mboxd.service: Unit entered failed state.
Jul 25 19:25:05 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
Jul 25 19:25:08 witherspoon systemd[1]: mboxd.service: Start request repeated too quickly.
Jul 25 19:25:08 witherspoon systemd[1]: Failed to start Phosphor MBOX Daemon.
Jul 25 19:25:08 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
```

- [x] Journald 
[journald_v1.99.8-175-g6c9f9d8_UBI_Error.txt](https://github.com/openbmc/openbmc/files/1195954/journald_v1.99.8-175-g6c9f9d8_UBI_Error.txt)

gkeishin, 2017-08-03:
And we couldnt powron as well  and subsequent Mbox services failing because of above due to crash

```
Jul 25 19:35:14 witherspoon systemd[1]: mboxd-reload@0.service: Main process exited, code=exited, status=255/n/a
Jul 25 19:35:14 witherspoon systemd[1]: Failed to start Reload mboxd during power on.
Jul 25 19:35:14 witherspoon systemd[1]: mboxd-reload@0.service: Unit entered failed state.
Jul 25 19:35:14 witherspoon systemd[1]: mboxd-reload@0.service: Failed with result 'exit-code'
```

geissonator, 2017-08-03:
Sounds like you did not update to the new squashfs.tar PNOR file?  That is required now when using latest BMC code.

closed 2017-08-03 by geissonator

gkeishin, 2017-08-03:
We still see this one also in https://github.com/openbmc/openbmc/issues/2083 log with new squase PNOR