openbmc/openbmc #2081: generate-squashfs: line 7: syntax error near unexpected token `newline'

PavamanSubramaniyam, 2017-08-03:
I have downloaded the generate-squashfs onto a P8 Machine with Ubuntu 16.04.2 OS from the link:
https://github.com/openbmc/openpower-pnor-code-mgmt/blob/master/generate-squashfs

Then tried to create the squashfs using the pnor image generated by op-build for witherspoon.
I am following the steps mentioned in https://gerrit.openbmc-project.xyz/#/c/4756/4/code-update.md

Generate a squashfs image. You'll need:
  * A PNOR image: https://github.com/open-power/op-build
  * The generate-squashfs script:
  https://github.com/openbmc/openpower-pnor-code-mgmt/blob/master/generate-squashfs
  * Run: `generate-squashfs <pnor-image>`

```
root@powerkvm3-lp1:~# ./generate-squashfs /root/P9_witherspoon/op-build/output/images/witherspoon-redbud.pnor
./generate-squashfs: line 7: syntax error near unexpected token `newline'
./generate-squashfs: line 7: `<!DOCTYPE html>'
```

PavamanSubramaniyam, 2017-08-03:
I have generated the witherspoon-redbud.pnor image using the latest op-build git tree.
Wanted to check the new method of flashing the squashfs onto my Open BMC witherspoon hardware.

williamspatrick, 2017-08-03:
The user appears to have downloaded an HTML rendering of the script from github rather than the raw file. The real script does not have DOCTYPE HTML tags in it.

closed 2017-08-03 by williamspatrick

mdmillerii, 2017-08-03:
A better link uses raw instead of blob in the url 

https://github.com/openbmc/openpower-pnor-code-mgmt/raw/master/generate-squashfs 

which gets redirected to 

https://raw.githubusercontent.com/openbmc/openpower-pnor-code-mgmt/master/generate-squashfs