openbmc/openbmc #2083: BMC not booting after new way of pnor update

prkatti1, 2017-08-03:
We followed these steps:

Curl Commands:
export BMC_IP=<your bmc ip>
curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" -X POST -T <path of pnor squashfs.tar file> https://$BMC_IP/upload/image
curl -c cjar -b cjar -k -H "Content-Type: application/json" https://$BMC_IP:<port>/xyz/openbmc_project/software/enumerate
curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":"xyz.openbmc_project.Software.Activation.RequestedActivations.Active"}' https://$BMC_IP/xyz/openbmc_project/software/<image id>/attr/RequestedActivation
1. Initial Test - Moving from old bmc code to new bmc code that supports new PNOR update

Update BMC
Verify BMC Ready (obmcutil state)
Upload new PNOR
Activate new PNOR
Reboot BMC
Verify BMC Ready
(Run workarounds.sh) -> Workaround to max chip issue
Boot system, verify host boots
2. New PNOR Test - Assumes first test complete, upload new PNOR image while at BMC Ready, System Off

Power off system
Upload new PNOR
Activate new PNOR
(Run workarounds.sh) -> Workaround to max chip issue
Boot system, verify host boots

It is not booting.

Version Info:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-175"
VERSION_ID="v1.99.8-175-g6c9f9d8"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-175"
BUILD_ID="v1.99.8"

CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

prkatti1, 2017-08-03:
[journalogs.txt](https://github.com/openbmc/openbmc/files/1197006/journalogs.txt)

geissonator, 2017-08-03:
The workarounds.sh is not longer needed (and may actually be bad for things) with the latest master.

gkeishin, 2017-08-03:
```
Jul 25 19:24:55 witherspoon mboxd[1341]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 25 19:24:55 witherspoon systemd[1]: mboxd.service: Main process exited, code=dumped, status=6/ABRT
Jul 25 19:24:55 witherspoon mboxctl[1342]: Failed to post message: No route to host
Jul 25 19:24:55 witherspoon mboxctl[1342]: Failed to send reset command
Jul 25 19:24:55 witherspoon mboxd[1341]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Jul 25 19:24:55 witherspoon systemd[1]: mboxd.service: Unit entered failed state.
Jul 25 19:24:55 witherspoon mboxd[1341]: [ 1501010695.241380217] Partition HBB not found
Jul 25 19:24:55 witherspoon systemd[1]: mboxd.service: Failed with result 'core-dump'.
Jul 25 19:24:55 witherspoon systemd[1]: mboxd-reload@0.service: Main process exited, code=exited, status=255/n/a
Jul 25 19:24:55 witherspoon systemd[1]: Failed to start Reload mboxd during power on.
Jul 25 19:24:55 witherspoon systemd[1]: mboxd-reload@0.service: Unit entered failed state.
Jul 25 19:24:55 witherspoon systemd[1]: mboxd-reload@0.service: Failed with result 'exit-code'.
Jul 25 19:24:55 witherspoon systemd[1]: Started power on mode unbind ir35221 device driver devs.
Jul 25 19:24:56 witherspoon phosphor-gpio-monitor[1332]: GPIO line altered
Jul 25 19:24:56 witherspoon kernel: UBIFS error (pid: 1195): cannot open "ubi7:pnor-patch", error -19
Jul 25 19:24:56 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 25 19:24:56 witherspoon kernel[1143]: [   39.140000] UBIFS error (pid: 1195): cannot open "ubi7:pnor-patch", error -19
```

prkatti1, 2017-08-03:
pflash -r /tmp/out.txt -P VERSION
FFS: Reading the flash has returned all 0xFF.
     Are you reading erased flash?
     Is something else using the flash controller?
Error 15 opening ffs !

df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     215216         0    215216   0% /dev
tmpfs                   216132     63552    152580  29% /run
/dev/mtdblock4           14464     14464         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       560      3536  14% /run/initramfs/rw
cow                       4096       560      3536  14% /
tmpfs                   216132         0    216132   0% /dev/shm
tmpfs                   216132         0    216132   0% /sys/fs/cgroup
tmpfs                   216132         0    216132   0% /tmp
tmpfs                   216132       404    215728   0% /var/volatile
tmpfs                   216132         0    216132   0% /usr/local

geissonator, 2017-08-03:
pflash no longer works with the new PNOR, @anoo1 can provide command to read the version info now.

saqibkh, 2017-08-03:
Prashanth probably rebooted the BMC while the PNOR Code Update was in progress. Therefore there were no ubi volumes on the system, nor was the pnor chip formatted as ubi.

I executed another PNOR code update on that system and we were able to boot the HOST successfully.

prkatti1, 2017-08-03:
Yes, but taking lil longer time to activate.

saqibkh, 2017-08-06:
@rfrandse Please close this issue as there are no action items here.

closed 2017-08-07 by geissonator