openbmc/openbmc #2086: [xCAT] Takes 2 delete actions to remove PNOR image in Ready state.

gurevichmark, 2017-08-03:
Running on 08/03 daily build.

If PNOR image is in "Active" state, the `action/Delete` removes the image as expected. 
But if image is in "Ready" state, takes 2 Delete calls to remove it.

1. Upload a PNOR image.
2. enumerate will show:
```
"/xyz/openbmc_project/software/32f129cd": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Ready",
      "ExtendedVersion": "op-build-v1.17-177-gdd8f32b-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2-p7582c9b,hostboot-3fa8be9,linux-4.12.1-openpower1-p899da6c,petitboot-v1.5.0-pc8fef8e,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p056301c",
      "Path": "/tmp/images/32f129cd",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.68"
    }
```
3. Run delete action:
```
# curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST  -d '{"data": []}' https://$BMCIP/xyz/openbmc_project/software/32f129cd/action/Delete
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
4. enumerate will show the image still there but with fewer attributes displayed:
```
"/xyz/openbmc_project/software/32f129cd": {
      "Path": "/tmp/images/32f129cd",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.68"
    },
```
5. Run delete action again. This time the image object is gone.

lgon, 2017-08-03:
I recreated the delete issue with some new log messages so I could trace what was happening.  I sent  a delete curl command for a pnor image in Ready state and only the Version object in the org.open_power.Software.Host.Updater.service received the command.  When sent a second time, the image manager in xyz.openbmc_project.Software.Version.service received it.

`
PNOR Image in "Ready State"
After first delete:
1.journalctl -u org.open_power.Software.Host.Updater.service
-----------
Jul 25 23:52:07 witherspoon systemd[1]: Started OpenPower Software Update Manager.
Jul 25 23:55:05 witherspoon openpower-update-manager[5624]: [[0;1;31mError in reading file[[0m
Jul 25 23:55:35 witherspoon openpower-update-manager[5624]: [[0;1;31mLEO-----------------  PNOR ItemUpdater::erase() - removed partitions[[0m
Jul 25 23:55:36 witherspoon openpower-update-manager[5624]: [[0;1;31mLEO-----------------  PNOR ItemUpdater::erase() - removed from versions[[0m
Jul 25 23:55:36 witherspoon openpower-update-manager[5624]: [[0;1;31mLEO-----------------  PNOR ItemUpdater::erase() - removed from activations[[0
m
2.journalctl -u xyz.openbmc_project.Software.Version.service
------------
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mError No manifest file[[0m
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mAn error when reading the Manifest file.[[0m
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mError processing image[[0m
Jul 25 23:55:04 witherspoon phosphor-version-software-manager[5632]: Untaring

3. journalctl -u xyz.openbmc_project.Software.BMC.Updater.service
-----------
Jul 25 23:52:08 witherspoon systemd[1]: Started OpenBMC Software Update Manager.

~
After second delete
1.journalctl -u org.open_power.Software.Host.Updater.service
-----------
No change
2.journalctl -u xyz.openbmc_project.Software.Version.service
------------
Jul 25 23:52:07 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Jul 25 23:52:07 witherspoon systemd[1]: Started Phosphor Version Software Manager.
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: tar: short read
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mError No manifest file[[0m
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mAn error when reading the Manifest file.[[0m
Jul 25 23:54:55 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mError processing image[[0m
Jul 25 23:55:04 witherspoon phosphor-version-software-manager[5632]: Untaring
Jul 26 00:01:35 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mLEO--------------- BMC Manager::erase()[[0m
Jul 26 00:01:35 witherspoon phosphor-version-software-manager[5632]: [[0;1;31mLEO--------------- BMC Manager::erase Deleting dir()[[0m
`

lgon, 2017-08-03:
I'm going to check rest_dbus.py to see if it's only sending the delete command to the first service that has a matching dbus destination.

lgon, 2017-08-08:
If xyz.openbmc_project.Software.Version.service is started first, and then org.open_power.Software.Host.Updater.service, the first delete command will delete the partitions, and the second delete command will delete the image dir. If Host.Updater is started first, the dir is deleted first. (e

gurevichmark, 2017-08-14:
Sorry, little confused by the estimate of "0". Will that be fixed in some future build before September release ?

eddiejames, 2017-08-25:
Task List

- [ ] Remove image updater delete method
- [ ] Test

# **STATUS**
08/24 - Familiarized with phosphor-bmc-code-mgmt, coded up a solution. Need to test

rfrandse, 2017-08-30:
<https://gerrit.openbmc-project.xyz/6316> Fix problem requiring two Delete calls to remove an image
Resolves  #2086 [xCAT] Takes 2 delete actions to remove PNOR image in Ready state.

gurevichmark, 2017-09-11:
Works for me now.

rfrandse, 2017-09-11:
Eddie James
4:09 PM
hm, dunno. surprised it works for him without my fix…
Adriana Kobylak
4:10 PM
i built a test image for him with your commit
Eddie James
4:10 PM
oh, cool, there we go
Reed Frandsen
4:10 PM
ok I'll reopen for tracking purpose

closed 2017-09-12 by williamspatrick