openbmc/openbmc #2087: Setting networks to DHCP from Petiteboot, isn't preserved when Witherspoon is rebooted

moralesj, 2017-08-03:
Trying to provision a Witherspoon, using xCat.  Petiteboot is showing a network interface with a static IP address.  

Networks are set back to DHCP using Petiteboot System Menu, and selecting DHCP for all devices., and then selecting OK.  When the system is rebooted, and the networks are check in Petiteboot, the devices show that they have reverted back to having a static IP, which is incorrect.

sammj, 2017-08-03:
This is a duplicate of https://github.com/openbmc/openbmc/issues/2033

ratagupt, 2017-08-04:
I am not sure it is a duplicate to 2033
I need below info.
1) What was the ipmi command which you ran?
2) Did you verify that network settings get applied on openbmc?
3) I assume that as ipmitool doesn't throw error,so it gets you a feeling that changes successfully applied.
For me that could be a settings validation issue.

sammj, 2017-08-04:
> Networks are set back to DHCP using Petiteboot System Menu, and selecting DHCP for all devices., and then selecting OK. When the system is rebooted, and the networks are check in Petiteboot, the devices show that they have reverted back to having a static IP, which is incorrect.

What Static IP is set? If it is `0.0.0.0`, then it will be a dupe of https://github.com/openbmc/openbmc/issues/2033

ratagupt, 2017-08-04:
no,it can't be set as settings validation doesn't allow that.

sammj, 2017-08-04:
@ratagupt Except as seen in https://github.com/openbmc/openbmc/issues/2033 that address _is_ getting set, because that's the default IP value in the OpenBMC code.

ratagupt, 2017-08-04:
yes.

ratagupt, 2017-08-04:
The below command works fine.

dkodihal, 2017-08-04:
I have a couple of fixes in the settings app that will definitely be required to solve this bug, but at the moment I don't know if networkd or host-ipmid needs changes as well. @ratagupt to confirm.

Here are my fixes (they're up for code review) :

https://gerrit.openbmc-project.xyz/#/c/5775/
https://gerrit.openbmc-project.xyz/#/c/5776/

ratagupt, 2017-08-04:
I am also hoping with these fixes,the issue should be fixed.
Let submitter to respond back my querys.

ralphbellofatto, 2017-08-04:
When can we expect to see a firmware update that includes this and when will it be incorporated in new witherspoon systems?

dkodihal, 2017-08-08:
This should be fixed in https://github.com/openbmc/openbmc/commit/201bd626032b334d1fee213be4272c1155662ae8.

closed 2017-08-08 by dkodihal