openbmc/openbmc #2088: Sprint 17.08.09 - Committed

rfrandse, 2017-08-03:


dkodihal, 2017-08-07:
***Squad 1 sprint plan***

**Ratan**
Defects
- #1533 BuildDate not getting updated under motherboard fru record 

**Devender**
- #1786 Define object specific inventory interfaces
- #1704 Write a bbclass to simplify copying of error yaml files in native recipes

**Vishwa**
- #1714 REST API to set a user login password
Defects
- #2024 openpower-occ-control crashes on single socket system

**Tom**
- #1671 Phosphor-networkd:- Use case validation
Defects
- #2101 RH1469630- power9: witherspoon console SOL session displays msgs

**Jayanth**
- #2078 Enable dump collection support for Internal error

**Deepak**
- #1856 Design event logging mechanism
- #1966 External Event Notification
Defects
- #1689 [Witherspoon]:REST Logging attribute "Resolved" can't be set from external

**Dhruv**

anoo1, 2017-08-08:
# Squad 3 -

**Adriana:anoo1 -**
**17.07.12**
-  #1940 BMC: Ability for u-boot to execute the kernel that's in a ubi volume
- #1943 BMC: Tell u-boot where squashfs image is
---
**Saqib:saqibkh -**
**17.08.09**
- #2113 BMC Update: Verification, test and automation integration
---
**Michael:mtritz -**
**17.08.09**
- #2007 ItemUpdater: Implement Associations
---
**Charlie: -** 
**17.08.09**
---
**Leo:lgon -**
**17.07.26**
- #1663 Rename mboxd dbus object to be in correct xyz.openbmc_project
**17.08.09**
- #2072 Make /media tmpfs
---

geissonator, 2017-08-09:
# Squad 2 -

 **Andrew:geissonator -**

**17.07.26:**

- #2032 (host reboot followed by chassis off results in reboot)
   * 08/09: ready for Patrick

- #1593 (Soft Power Off by pressing power button broken)
   * 08/09: ready for Patrick

**17.08.09:**

- #2119 (Manual scan of fsi slaves)

- #1676 (reboot count decrement)

- RAS error inject defect investigation

- https://gerrit.openbmc-project.xyz/#/c/3749/ (old inventory removal)

---

**Gunnar:gtmills:**

**17.07.12:**

- #1860 (PDM elogs)
  * 07/31: dug into PDM
  * 08/01: started coding, new class elog
  * 08/02: created new errors, continued on coding

---

**Chris:cbostic:**

**17.07.12:**

- #1727 (ucd driver support)
  * 08/01: worked on
  * 08/02: working on
  * 08/07: hit problems but can continue

**17.08.09:**

---

**Brandon:bjwyman:**

**17.07.26:**

- #1731 (power supply monitor for faults)
  * 07/31: Will get with Barth
  * 08/01: Will need device driver

**17.08.09:**

---

**Eddie:eddiejames:**

**17.06.28:**

- #1883 (hwmon skeleton for power supply driver)
  * 07/31: Having some issues on the different systems

**17.08.09:**

- #1821 (occ throttling and other event sensors in hwmon)

---

**Matt S:spinler:**

**17.06.28:**

- #1724 (chip fault analysis)
  * 08/01: Working on
  * 08/02: Design topic
  * 08/07: found a way around problem

**17.08.09:**

---

**Matt B:msbarth:**

**17.06.28:**

- #1835 (delay on events parameter)
  * 07/27: Have a good idea of layout
  * 07/31: code finished
  * 08/01: will wrap up today
  * 08/02: hit problems, hope to figure out today

- #1839 (handle scale property)

Lower priority, finish as time allows

- #1792 (yaml updates)

- Focus on smart fan speed integration, need to ensure we don't destabilize witherspoon beta

**17.08.09:**

---

**Brad:bradbishop:**

**17.08.09:**

---

closed 2017-08-31 by rfrandse