openbmc/openbmc #2095: [xCAT] Priority attribute not displayed for BMC image after a reboot.

gurevichmark, 2017-08-04:
Running on a 08/03 daily build.

After BMC reboot, `/software/enumerate` displays Host and BMC images, but BMC image is missing a "Priority" attribute:

```
#curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://$BMCIP/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/2a1022fe": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.17-176-g11dd356-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2,hostboot-3fa8be9,linux-4.12.1-openpower1-p3a0d8f7,petitboot-v1.5.0-p3eae874,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p1db02ba",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    },
    "/xyz/openbmc_project/software/54c1c9c": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-178-g7a75570"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

saqibkh, 2017-08-04:
Did BMC image ever had a priority?

We haven't delivered BMC Code Update yet, so why are we getting these issues?

mtritz, 2017-08-04:
BMC priority persistence isn't implemented yet.

gurevichmark, 2017-08-14:
When can I expect the build where BMC priority persistence will be implemented ?

saqibkh, 2017-08-14:
The code is ready for merge, so most likely this week. Let me know if you need it for testing and I can create an image for your testing.

rfrandse, 2017-08-24:
#2095 is a dup of https://github.com/openbmc/openbmc/issues/2137
and is resolved by https://gerrit.openbmc-project.xyz/#/c/5946/
Closing this as a dup

closed 2017-08-24 by rfrandse