openbmc/openbmc #2098: dependency issue on bmc reboots - ordering cycle on local-fs.target/stop

geissonator, 2017-08-04:
Saw this with latest master (9b3beee10191fbec68345ec48829f7bc3728c576) on BMC reboots
```
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found ordering cycle on local-fs.target/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on usr-local-share-pnor.mount/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on usr-local.mount/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on org.open_power.Software.Host.Updater.service/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on basic.target/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on sockets.target/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on rest-dbus.socket/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on sysinit.target/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on systemd-update-done.service/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Found dependency on local-fs.target/stop
Aug 04 19:11:16 witherspoon systemd[1]: local-fs.target: Breaking ordering cycle by deleting job usr-local-share-pnor.mount/stop
Aug 04 19:11:16 witherspoon systemd[1]: usr-local-share-pnor.mount: Job usr-local-share-pnor.mount/stop deleted to break ordering cycle starting with local-fs.target/stop
```

```
[witherspoon]# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     215220         0    215220   0% /dev
tmpfs                   216132     14436    201696   7% /run
/dev/mtdblock4           14464     14464         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       296      3800   7% /run/initramfs/rw
cow                       4096       296      3800   7% /
tmpfs                   216132         0    216132   0% /dev/shm
tmpfs                   216132         0    216132   0% /sys/fs/cgroup
tmpfs                   216132         0    216132   0% /tmp
tmpfs                   216132       172    215960   0% /var/volatile
ubi7:pnor-rw-e49bc78e
                         13816        16     13056   0% /media/pnor-rw-e49bc78e
/dev/ubiblock7_1         19456     19456         0 100% /media/pnor-ro-e49bc78e
ubi7:pnor-prsv             972        16       884   2% /media/pnor-prsv
tmpfs                   216132         0    216132   0% /usr/local
ubi7:pnor-patch          13816        16     13056   0% /usr/local/share/pnor
```

geissonator, 2017-08-04:
Phase 5 until we know root cause and if we can defer.

rfrandse, 2017-08-08:
<https://gerrit.openbmc-project.xyz/5849> Resolves the dependency issue on usr-local.mount
Resolves  #2098 dependency issue on bmc reboots - ordering cycle on local-fs.target/stop

closed 2017-08-14 by williamspatrick