openbmc/openbmc #2100: Host.HostState.Quiesced and machine keeps on doing ISTEPs

PavamanSubramaniyam, 2017-08-07:
I have a P9 Witherspoon machine with latest Open BMC and PNOR Firmware images.
I am trying to power on the HOST using obmcutil poweron command from BMC ssh login.

```
root@witherspoon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced

--== Welcome to Hostboot  ==--

  4.05968|secure|Booting in non-secure mode.
  4.05970|secure|SecureROM invalid - skipping functionality
  6.32097|ERRL|Dumping errors reported prior to registration
  7.30761|ISTEP  6. 5 - host_init_fsi
  7.39349|ISTEP  6. 6 - host_set_ipl_parms
  7.98138|ISTEP  6. 7 - host_discover_targets
  8.53122|HWAS|PRESENT> DIMM[03]=FFFF000000000000
  8.53123|HWAS|PRESENT> Proc[05]=C000000000000000
  8.71745|ISTEP  6. 8 - host_update_master_tpm
  8.73269|SECURE|Security Access Bit> 0x0000000000000000
  8.73270|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
  8.73290|ISTEP  6. 9 - host_gard
  8.78967|ISTEP  6.10 - host_revert_sbe_mcs_setup
  8.80130|ISTEP  6.11 - host_start_occ_xstop_handler
  8.80220|ISTEP  6.12 - host_voltage_config
  8.83754|ISTEP  7. 2 - mss_volt
  9.14854|ISTEP  7. 3 - mss_freq
  9.27121|ISTEP  7. 4 - mss_eff_config
 12.53811|ISTEP  7. 5 - mss_attr_update
 12.55075|ISTEP  8. 1 - host_slave_sbe_config
 12.59290|ISTEP  8. 2 - host_setup_sbe
 12.59905|ISTEP  8. 3 - host_cbs_start
 12.60705|ISTEP  8. 4 - proc_check_slave_sbe_seeprom_complete
 21.02141|ISTEP  8. 5 - host_attnlisten_proc
 21.02402|ISTEP  8. 6 - host_p9_fbc_eff_config
 21.03104|ISTEP  8. 7 - host_p9_eff_config_links
 21.04055|ISTEP  8. 8 - proc_attr_update
 21.04243|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 21.09585|ISTEP  8.10 - proc_xbus_scominit
 24.32066|ISTEP  8.11 - proc_xbus_enable_ridi
 24.32525|ISTEP  9. 1 - fabric_erepair
 24.33572|ISTEP  9. 2 - fabric_io_dccal
 25.19470|ISTEP  9. 3 - fabric_pre_trainadv
 25.20075|ISTEP  9. 4 - fabric_io_run_training
 25.41569|ISTEP  9. 5 - fabric_post_trainadv
 25.42053|ISTEP  9. 6 - proc_smp_link_layer
 25.42567|ISTEP  9. 7 - proc_fab_iovalid
 25.45104|ISTEP  9. 8 - host_fbc_eff_config_aggregate
 25.46131|ISTEP 10. 1 - proc_build_smp
 25.77868|ISTEP 10. 2 - host_slave_sbe_update
 26.06887|ISTEP 10. 3 - host_set_voltages
 26.08340|ISTEP 10. 4 - proc_cen_ref_clk_enable
 26.08933|ISTEP 10. 5 - proc_enable_osclite
 26.09030|ISTEP 10. 6 - proc_chiplet_scominit
 26.12315|ISTEP 10. 7 - proc_abus_scominit
 26.12922|ISTEP 10. 8 - proc_obus_scominit
 26.21275|ISTEP 10. 9 - proc_npu_scominit
 26.24961|ISTEP 10.10 - proc_pcie_scominit
 26.30546|ISTEP 10.11 - proc_scomoverride_chiplets
 26.30744|ISTEP 10.12 - proc_chiplet_enable_ridi
 26.31142|ISTEP 10.13 - host_rng_bist
 26.31639|ISTEP 10.14 - host_update_redundant_tpm
 26.37564|ISTEP 11. 1 - host_prd_hwreconfig
 26.38136|ISTEP 12. 1 - mss_getecid
 26.38809|ISTEP 13. 1 - host_disable_memvolt
 26.50477|ISTEP 13. 2 - mem_pll_reset
 26.51853|ISTEP 13. 3 - mem_pll_initf
 26.55219|ISTEP 13. 4 - mem_pll_setup
 26.57987|ISTEP 13. 6 - mem_startclocks
 26.60065|ISTEP 13. 7 - host_enable_memvolt
 26.60266|ISTEP 13. 8 - mss_scominit
 27.10196|ISTEP 13. 9 - mss_ddr_phy_reset
 27.22425|ISTEP 13.10 - mss_draminit
 27.48427|ISTEP 13.11 - mss_draminit_training
 28.12494|ISTEP 13.12 - mss_draminit_trainadv
 28.12817|ISTEP 13.13 - mss_draminit_mc
 28.15800|ISTEP 14. 1 - mss_memdiag
 39.10365|ISTEP 14. 2 - mss_thermal_init
 39.11315|ISTEP 14. 3 - proc_pcie_config
 39.12718|ISTEP 14. 4 - mss_power_cleanup
 39.13053|ISTEP 14. 5 - proc_setup_bars
 39.17614|ISTEP 14. 6 - proc_htm_setup
 39.18722|ISTEP 14. 7 - proc_exit_cache_contained
 39.20736|ISTEP 15. 1 - host_build_stop_image
 39.57546|ISTEP 15. 2 - proc_set_pba_homer_bar
 39.58524|ISTEP 15. 3 - host_establish_ex_chiplet
 39.59571|ISTEP 15. 4 - host_start_stop_engine
 39.70169|ISTEP 16. 1 - host_activate_master
  0.00632|ISTEP 16. 2 - host_activate_slave_cores
 40.84830|ISTEP 16. 3 - host_secure_rng
 40.85138|ISTEP 16. 4 - mss_scrub
 40.88613|ISTEP 16. 5 - host_ipl_complete
 41.21072|ISTEP 16. 6 - host_load_io_ppe
 41.24334|ISTEP 18.11 - proc_tod_setup
 41.26901|ISTEP 18.12 - proc_tod_init
 41.30775|ISTEP 20. 1 - host_load_payload
 41.71105|ISTEP 20. 2 - host_load_hdat
 45.61534|ISTEP 21. 1 - host_runtime_setup
```

PavamanSubramaniyam, 2017-08-07:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-167"
VERSION_ID="v1.99.8-167-g3ab09ac"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-167"
BUILD_ID="v1.99.8"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x0280d000..0x0280e000 !
[                                                  ] 0%open-power-witherspoon-v1.18-32-g21dbb70
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-24-g81c0e076374e-pc665224
        hostboot-faac61e
        linux-4.12.3-openpower1-p3f3388a
        petitboot-v1.5.1-pb361a08
        machine-xml-5575e7e
        occ-d650501
        hostboot-binaries-883d218
        capp-ucode-9c73e9f
        sbe-02021c6-p757955a
[==================================================] 100%
```

PavamanSubramaniyam, 2017-08-07:
[20170807065751797890_Myffdc.zip](https://github.com/openbmc/openbmc/files/1204080/20170807065751797890_Myffdc.zip)

geissonator, 2017-08-18:
This is a host checkstop, I know they've put a few fixes in recently for this.  This looks like the occ reset checkstop.  Hostboot put a fix in for this I believe.  These types of issues should be opened on the op-build github repo.
```
Aug 07 06:17:29 witherspoon checkstop_app[2012]: Host checkstop condition detected
Aug 07 06:17:29 witherspoon phosphor-fru-fault-monitor[1039]: /xyz/openbmc_project/logging/entry/2
Aug 07 06:17:29 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Aug 07 06:17:29 witherspoon systemd[1]: Reached target Quiesce Target.
Aug 07 06:17:29 witherspoon systemd[1]: Reached target Host instance 0 crashed.
Aug 07 06:17:30 witherspoon phosphor-host-state-manager[1068]: Auto reboot enabled. Reducing HOST BOOTCOUNT by 1.
Aug 07 06:17:30 witherspoon phosphor-host-state-manager[1068]: Beginning reboot...
Aug 07 06:17:30 witherspoon phosphor-host-state-manager[1068]: Host State transaction request
```

closed 2017-08-18 by geissonator