openbmc/openbmc #2109: PNOR code update: Mounting squashfs fails, causing a core dump

mtritz, 2017-08-07:
Caught by Robot tests in latest code update build.
```
Aug 03 15:51:15 witherspoon systemd[1]: Starting Mount SquashFS image pnor-ro-e49bc78e...
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: attaching mtd7
Aug 03 15:51:17 witherspoon obmc-flash-bios[1232]: Volume ID 1, size 257 LEBs (16809856 bytes, 16.0 MiB), LEB size 65408 bytes (63.9 KiB), dynamic, name "pnor-rw-e49bc78e", alignment 1
Aug 03 15:51:17 witherspoon obmc-flash-bios[1233]: Volume ID 0, size 385 LEBs (25182080 bytes, 24.0 MiB), LEB size 65408 bytes (63.9 KiB), static, name "pnor-ro-e49bc78e", alignment 1
Aug 03 15:51:16 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@e49bc78e.service: Main process exited, code=exited, status=32/n/a
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: scanning is finished
Aug 03 15:51:17 witherspoon obmc-flash-bios[1233]: ubiupdatevol: error!: cannot start volume "/dev/ubi7_0" update
Aug 03 15:51:17 witherspoon obmc-flash-bios[1233]:               error 16 (Device or resource busy)
Aug 03 15:51:16 witherspoon systemd[1]: Failed to start Mount SquashFS image pnor-ro-e49bc78e.
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: attached mtd7 (name "pnor", size 128 MiB)
Aug 03 15:51:17 witherspoon obmc-flash-bios[1233]: mount: /dev/ubiblock7_0: can't read superblock
Aug 03 15:51:16 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@e49bc78e.service: Unit entered failed state.
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
Aug 03 15:51:17 witherspoon obmc-flash-bios[1233]: /usr/sbin/obmc-flash-bios: error 32
Aug 03 15:51:16 witherspoon systemd[1]: obmc-flash-bios-squashfsmount@e49bc78e.service: Failed with result 'exit-code'.
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: min./max. I/O unit sizes: 1/256, sub-page size 1
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: VID header offset: 64 (aligned 64), data offset: 128
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: good PEBs: 2048, bad PEBs: 0, corrupted PEBs: 0
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: user volume: 0, internal volumes: 1, max. volumes count: 128
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: max/mean erase counter: 0/0, WL threshold: 4096, image sequence number: 1833331633
Aug 03 15:51:15 witherspoon kernel[956]: [  507.220000] ubi7: available PEBs: 2042, total reserved PEBs: 6, PEBs reserved for bad PEB handling: 0
Aug 03 15:51:16 witherspoon kernel[956]: [  508.130000] ubi7: background thread "ubi_bgt7d" started, PID 1231
Aug 03 15:51:16 witherspoon kernel[956]: [  508.130000] block ubiblock7_0: created from ubi7:0(pnor-ro-e49bc78e)
Aug 03 15:51:17 witherspoon kernel[956]: [  509.040000] ubi7 error: get_exclusive: 2 users for volume 0
Aug 03 15:51:17 witherspoon kernel[956]: [  509.040000] blk_update_request: I/O error, dev ubiblock7_0, sector 0
Aug 03 15:51:17 witherspoon kernel[956]: [  509.040000] SQUASHFS error: squashfs_read_data failed to read block 0x0
Aug 03 15:51:17 witherspoon kernel[956]: [  509.040000] squashfs: SQUASHFS error: unable to read squashfs_super_block
```
snip
```
Aug 03 15:51:18 witherspoon kernel[956]: [  511.110000] UBIFS (ubi7:2): reserved for root: 71055 bytes (69 KiB)
Aug 03 15:51:18 witherspoon kernel[956]: [  511.110000] UBIFS (ubi7:2): media format: w5/r0 (latest is w5/r0), UUID C6B3DB0F-0135-47B6-A89C-97DFD32A72CB, small LPT model
Aug 03 15:51:18 witherspoon systemd[1]: Started Mount UBIFS volume pnor-prsv and pnor-rw-e49bc78e.
Aug 03 15:51:18 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=dumped, status=11/SEGV
Aug 03 15:51:18 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Aug 03 15:51:18 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'core-dump'.
```

mtritz, 2017-08-08:
Couldn't recreate - neither could subsequent CT builds, but did identify where the problem happens. Adding some extra traces in the event of another failure -
https://gerrit.openbmc-project.xyz/#/c/5855/
- and moving to backlog for now.

mdmillerii, 2017-08-11:
Actually I think I have an idea of how this might have failed.

The log shows the ubi device scan finding two volumes, 0 and 1.  The ubi block device is created first.  Adding the device likely triggers `udev` rules asynchronously causing the ubi volume to be opened.  Then `ubiupdatevol` is called to change the contents.  The return code `EBUSY` comes from a conflicting open, which is likely from the newly added block device alias.

This shows multiple issues:
1) the block device is created before the contents are set
2) content is being written to an existing volume

(1) If we are going to volume names based on content version then we should never allow a volume to be named without that content present.  Instead we should create a volume in a temp name, update the content, then rename the volume.  Using a temp name space means we can safely erase any such volumes upon a reboot.  (and parallel add will work if there is space, as assigning a volume to an existing name implicitly deletes the old named volume unless it is renamed in the same ioctl (utility invocation).

(2) We should not make the block device until after creating and naming the volume.

I'm guessing to recreate this one would need a volume to exist but not contain a vaild squashfs image.  Perhaps a prior reboot during activation.

mtritz, 2017-08-16:
Couldn't we just remove any existing volumes of the same name whenever we are creating a new volume during activation?

rfrandse, 2017-08-18:
<https://gerrit.openbmc-project.xyz/5855> PNOR code update: clear existing volumes before creating new ones
Resolves  #2109 PNOR code update: Mounting squashfs fails, causing a core dump

closed 2017-08-18 by williamspatrick