openbmc/openbmc #2110: Error on build from scratch?

anon99, 2017-08-07:
I tried building "bitbake obmc-phosphor-image" using a tree acquired using the zipfile and got this message, so I then git cloned the repo and have the same problem. [ImportError: No module found yaml]
in ...sdbusplus/1.0-r1/build/example]. I tied the workaround export PYTHONPATH but it didnt seem to help.
Does a "clean" build of HEAD really work?. Or is there some tweak/workaround (or something I'm missing.
[Somewhat similar to issue #846 thats closed].

mdmillerii, 2017-09-02:
Is your error similar to the error in #2196 ?

anon99, 2017-09-05:
In as much as it seems to revolve around:-
ImportError: No module found yaml

Yes!

gtmills, 2018-07-05:
#2196  is closed. Closing this issue. Please reopen if still an issue.

closed 2018-07-05 by gtmills