openbmc/openbmc #2111: IPL: Power off following Power On with phosphor-fan-control core dumps on wsbmc010 with 910.1732.20170804a

cdeadmin, 2017-08-07:


geissonator, 2017-08-07:
The error we log in phosphor-hwmon:
```
{
    "data": {
        "/xyz/openbmc_project/logging/entry/2": {
            "AdditionalData": [
                "CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a100.i2c-bus/i2c-3/3-0052",
                "CALLOUT_ERRNO=5"
            ],
            "Id": 2,
            "Message": "xyz.openbmc_project.Sensor.Device.Error.ReadFailure",
            "Resolved": 0,
            "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
            "Timestamp": 1501903549666,
            "associations": []
        }
    },
    "message": "200 OK",
    "status": "ok"
}
```

We just hit this on a system this morning.  Basically, if there any failure by phosphor-hwmon to read a sensor, it removes that sensor from it's dbus output permenantly.  If that sensor happens to be fan related, then the fan applications start failing.  The max chip seems to have this issue, we saw on wsbmc017 this morning.

If you look through the journal, you can see this:
```
Aug 05 03:25:45 witherspoon phosphor-hwmon-readd[981]: Failed to read from device.
``

Which maps to the max chip hwmon.
```
  981     1 root     S     6088   1%   0% /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/max31785@52
```

One proposed solution is to go back to old way of forcing phosphor-hwmon to just reboot itself when it hits a sensor read.  A more elegant solution is to retry failed reads for a bit before giving up.

pstrinkle, 2017-08-07:
On my unreliably test system, the phosphor-hwmon package would run out of reboot attempts from the read failures.  So, a solution that reboots the daemon on read failure will only last a few reboots before it's done and you're back where you were.  A solution I have use downstream is to always add the object to the dbus regardless of read failure and hard-coded in a failure value.  Obviously that last part isn't a great long term solution.  There have been discussions about setting the state of the sensors based on the success or failure of a sensor read.

rfrandse, 2017-08-10:
<https://gerrit.openbmc-project.xyz/5930> Configure hwmon to remove objects on fails
Resolves  #2111 IPL: Power off following Power On with phosphor-fan-control core dumps on wsbmc010 with 910.1732.20170804a

closed 2017-08-14 by williamspatrick