openbmc/openbmc #2115: hwmon config file for power supplies

spinler, 2017-08-08:
After the power supply driver is done, we need to make some hwmon config files to expose the power, current, and voltage readings for them.  

Required for early ship.

spinler, 2017-08-10:
- [ ] Create config files for each power supply with labels so all current, voltage, power, temp, and fan sensors get picked up by hwmon
- [ ] Check into the openbmc repository so they get installed with the conf files

spinler, 2017-08-10:
# **STATUS**
* 8/10: Put a preview of the PS driver on a system and built the config files and rebound the driver and saw everything show up in REST.

rfrandse, 2017-08-16:
<https://gerrit.openbmc-project.xyz/5982> Add power supply hwmon config files
Resolves  #2115 hwmon config file for power supplies

closed 2017-08-21 by williamspatrick